MarlinFirmware/Marlin #25866: Probe position causes slanting bed levelling [BUG] (bug summary)

fiveorders, 2023-05-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have tried from 2.0.9.3 to the latest nightly build. When I position the probe as ( 26, 0, 0 ) levelling is perfectly fine, but when I place the probe to ( 0, 49, 0 ) the bed starts to slant to the right side. the bed is 300x300, and the printer is a QGL like Voron.  The G34 works fine, bed levelling completes fine however when I start to print a 300x300 print it starts to scratch the bed gradually when moving to the right side. I have replaced the Octopus Pro board, and I have replaced the motors but nothing changed. I have tried with PINDA Probe and Genuine BL Touch and both resulted in the same. If I place the probe on the right side of the nozzle it works when I place it behind the nozzle it does this error.

### Bug Timeline

2.0.9.3

### Expected behavior

Print even

### Actual behavior

slanted towards the right side

### Steps to Reproduce

Place the probe at the back of the nozzle at a 49mm distance. Start a print which has a minimum 100-150mm square base

### Version of Marlin Firmware

2.1.2.1

### Printer model

Custom QGL Core XY

### Electronics

Octopus Pro, Bl Touch, PINDA Probe

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/11525402/Archive.zip)

thinkyhead, 2023-05-23:
What kind of result do you get if you do manual probing with either `PROBE_MANUALLY` or `MESH_BED_LEVELING`? If the result is better, this probably indicates there is an issue with the probe not staying perpendicular to the bed during probing. You can slso check to see if the result is better when you hold onto the wires and filament tube to prevent them from pulling on on the carriage as it moves. https://marlinfw.org/docs/basics/troubleshooting.html#one-side-high,-one-side-low

fiveorders, 2023-05-23:
Thank you for the response. I have yet to prob manually because I am still trying to figure out if I am missing something. The Probe is perfectly perpendicular to the nozzle and bed. And without changing anything (even the height of the probe), when I move to the right side of the extruder, everything is perfect. The result is the same with either the SuperPINDA or the BlTouch. I will flash klipper tonight and will check if it is okay on that.

closed 2023-05-24 by fiveorders