MarlinFirmware/Marlin #25868: Maple build environments are deprecated[BUG]

asdino2, 2023-05-22:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Sovol SV06 Plus  screen not respond

Warning also on Marlin 2.1.2.1

Marlin\src\inc\Warnings.cpp:770:4: warning: #warning "Maple build environments are deprecated. Please use a non-Maple build environment. Report issues to the Marlin Firmware project." [-Wcpp]
   #warning "Maple build environments are deprecated. Please use a non-Maple build environment. Report issues to the Marlin Firmware project."

### Expected behavior

Screen not comunicate 

### Version of Marlin Firmware

2.1.2.1

### Printer model

SOVOL SV06 PLUS

### Electronics

Stock

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/11527062/Configuration_adv.zip)

ellensp, 2023-05-22:
"Maple build environments are deprecated. Please use a non-Maple build environment."

Is a deliberate warning not to use maple environment where possible. IT IS NOT AN ERROR

ellensp, 2023-05-22:
there is no support for the this display yet, it is being worked on

https://github.com/MarlinFirmware/Marlin/pull/25864

closed 2023-05-22 by ellensp

asdino2, 2023-05-22:
Thank you for answer me. I will wait for #25864
With bugfix 2.1.x not working at all , other errors