MarlinFirmware/Marlin #25871: M109 delays printing on ender 3

tom-321, 2023-05-22:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When I start a print, Marlin waits a few seconds (sometimes almost a minute) before it starts (I guess it waits until the temperature is "stable" or something like that) 

I changed the temperature a little to make it visible:
[image]

It only happens with my Ender 3, the Artillery Sidewinder x2 starts immediately after reaching the set temperature 

My suggestion would be to give M109 an additional parameter that allows to set the temperature that must be reached for the wait sequence in addition to the total target temperature.

### Version of Marlin Firmware

Marlin-2.0.1 

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Archiv.zip](https://github.com/MarlinFirmware/Marlin/files/11537122/Archiv.zip)

thinkyhead, 2023-05-30:
There are configuration parameters to set the "Residency Time" if you would prefer a shorter wait period for temperature stabilization. You can also set a wider Hysteresis value if yiu want to consider a wider range of temperature "close enough." You can also change your G-code to first set a lower temperature and then set a non-wait temperature target that will apply to the rest of the print job. It can also help to tune your PID or use `MPCTEMP` which tends to stabilize more quickly.

tom-321, 2023-05-30:
Hi thanks,
that's nice 👍  I'll change it in the firmware (guess it's the TEMP_RESIDENCY_TIME setting which is set to 10) 
thought it was a gcode thing, but if there is a setting for it, it's allright for me :)

closed 2023-05-30 by tom-321