MarlinFirmware/Marlin #25879: [FR] (feature summary)

AeroSonic4, 2023-05-25:
### Is your feature request related to a problem? Please describe.

I need to have separate accelerations for the x and y axis (someone may need z), because most bed slingers need different values.  There are separate values for X and y speeds and for jerk; however, there isn't separate accelerations.  

### Describe the feature you want

If I may suggest add to the M204 command a PX, PY, and PZ for the X, Y, and Z accelerations, respectively,  Also keep P as it is today for backward compatibility.

The suggest command would look like: M204  [P<accel>] [PX<x-accel>] [PY<y-accel>] [PZ<z-accel>] [R<accel>] [S<accel>] [T<accel>]

thinkyhead, 2023-05-30:
While the base acceleration is set based on move type, and this necessarily applies to all axes involved in a linear move, there are in fact separate Max Acceleration limits for X and Y. And this can be set as part of a sliced print job, if an X-only or Y-only move needs special consideration. So no new feature is needed.

AeroSonic4, 2023-06-01:
I'm not sure I understand, as I have not been able to find the command in Marlin that sets the maximum acceleration limits independently for X and Y.  The only thing I have found in Marlin is the M204 command that sets the maximum acceleration as one value for both X and Y.  The M205 command has maximum feed velocities and jerks which are independent for the X and Y directions.  If you are saying this is such an acceleration command could you please help me identify it?  Thanks.

ellensp, 2023-06-01:
/**
 * Default Max Acceleration (speed change with time) (linear=mm/(s^2), rotational=°/(s^2))
 * (Maximum start speed for accelerated moves)
 * Override with M201
 *                                      X, Y, Z [, I [, J [, K...]]], E0 [, E1[, E2...]]
 */
#define DEFAULT_MAX_ACCELERATION      { 3000, 3000, 100, 10000 }

https://marlinfw.org/docs/gcode/M201.html