MarlinFirmware/Marlin #25880: [FR] Add "keepout zones"

Tupson444, 2023-05-25:
### Is your feature request related to a problem? Please describe.

When a printer uses clams to hold the bed plate (or whatever other irregular "obstacle"), if not being careful the nozzle can hit the clams and potentially damage something.

### Describe the feature you want

It would be good to have "keepout zones" that prevent movement if requested GCODE coordinates are in specified zones (halts and displays error?), similar to RepRap firmware: https://reprap.org/wiki/G-code#M599:_Define_keepout_zone

These zones should be respected like software endstops, even when coordinates are offset by M206.

This could be defined in configuration.h, but also by GCODE (for easier change), e.g. M599 like in RepRap firmware. Besides X & Y, there should be Z height of the keepout zones (e.g. height of clams), so that movement is not prevented on higher Z.

### Additional context

Some slicers have something similar when generating GCODE file (Bambu Studio I think), and maybe others will implement it too.

However, I often use M206 command to offset the coordinates in order to print the same file on a different part of the bed. In this case it would be better if there was an option in firmware.

thisiskeithb, 2023-05-25:
We have `AVOID_OBSTACLES` to prevent collision with bed clips while probing, but nothing to prevent motion in those areas for other operations:

https://github.com/MarlinFirmware/Marlin/blob/a3eace117ced5ad72227046465a6d878f66261d9/Marlin/Configuration.h#L2101-L2104

Tupson444, 2023-05-25:
> We have `AVOID_OBSTACLES` to prevent collision with bed clips while probing, but nothing to prevent motion in those areas for other operations:
> 
> https://github.com/MarlinFirmware/Marlin/blob/a3eace117ced5ad72227046465a6d878f66261d9/Marlin/Configuration.h#L2101-L2104

Nice, it means that there is already a basis that can be expanded for other movement (I guess).

EvilGremlin, 2023-05-25:
> prevent movement if requested GCODE coordinates are in specified zones (halts and displays error?)

This require calculating whole gcode up to certain height with applied offsets and whatnot before start. Marlin have no basis for this at all. There won't be any common code with AVOID_OBSTACLES, because leveling moves are predetermined at compile time. Checking on the fly is much easier, but not really acceptable.

> I often use M206 command to offset the coordinates in order to print the same file on a different part of the bed

Seems to be very edge case, i doubt anyone will pick it up. This feature will take quite a lot of effort and time to implement for very small user base, potentially single digit. Maybe when there will be some wider user base request that require dry gcode processing...

Tupson444, 2023-05-25:
Good points @EvilGremlin , if there is not enough interest it's probably not worth the effort, but it would be nice as an extra safety (faster printers may suffer more damage when colliding with clips).