MarlinFirmware/Marlin #25882: [FR] Add to Edit Encoder Rate Multiplier in settings

classicrocker883, 2023-05-25:
### Is your feature request related to a problem? Please describe.

just wanted the ability to change the encoder rate to adjust it on the fly.

### Describe the feature you want

So I got this working!
I hope someone can review this code and give me some *pointers* (pun intended)

Now, the code may not be A+ 100% whatever typically should be, but it works!

<br>

File **lcd/e3v2/common/encoder.cpp** line ~130
```
    #if ENABLED(ENCODER_RATE_MULTIPLIER)
      int* encRateA = &ui.enc_rateA;
      int32_t* encRateB = &ui.enc_rateB;
      int a = *encRateA;
      int32_t b = *encRateB;
      millis_t ms = millis();
      int32_t encoderMultiplier = 1;
------------------[after lines of code]---------------
               if (encoderStepRate >= ENCODER_100X_STEPS_PER_SEC) encoderMultiplier = a;
          else if (encoderStepRate >= ENCODER_10X_STEPS_PER_SEC)  encoderMultiplier = b;
```

<br>

File **lcd/marlinui.cpp** line ~70
```
constexpr uint8_t epps = ENCODER_PULSES_PER_STEP;

#if ENABLED(ENCODER_RATE_MULTIPLIER)
  int MarlinUI::enc_rateA;
  int32_t MarlinUI::enc_rateB;
#endif
```

<br>

File **lcd/marlinui.cpp** line ~200
after `class MarlinUI {`
```
public:

  #if ENABLED(ENCODER_RATE_MULTIPLIER)
    static int enc_rateA;
    static int32_t enc_rateB;
  #endif
```

<br>

File **src/module/settings.cpp**
in the `SettingsDataStruct {`
```
  //
  // Encoder Rate
  //
  #if ENABLED(ENCODER_RATE_MULTIPLIER)
    int enc_rateA;
    int32_t enc_rateB;
  #endif
------------------[Many lines, goes with similar code]--------------
    #if ENABLED(ENCODER_RATE_MULTIPLIER)
      EEPROM_WRITE(ui.enc_rateA);
      EEPROM_WRITE(ui.enc_rateB);
    #endif
------------------[Many lines, goes with similar code]--------------
      #if ENABLED(ENCODER_RATE_MULTIPLIER)
        _FIELD_TEST(enc_rateA);
        EEPROM_READ(ui.enc_rateA);
        _FIELD_TEST(enc_rateB);
        EEPROM_READ(ui.enc_rateB);
      #endif
------------------[Many lines, goes with similar code]--------------
  #if ENABLED(ENCODER_RATE_MULTIPLIER)
    ui.enc_rateA = 550;
    ui.enc_rateB = 110;
  #endif
```

<br>

File **dwin.cpp**
```
void SetEncRate() { SetPIntOnClick( 0, 1000); }
-----------------[Many lines]--------------
    #if ENABLED(ENCODER_RATE_MULTIPLIER)
      EDIT_ITEM_F(ICON_ProbeMargin, "Enc Rate 1000x", onDrawPIntMenu, SetEncRate, &ui.enc_rateA);
      EDIT_ITEM_F(ICON_ProbeMargin, "Enc Rate 100x", onDrawPInt32Menu, SetEncRate, &ui.enc_rateB);
    #endif
-----------------[Include in a menu function]--------------
```

classicrocker883, 2023-05-25:
I wasn't sure having a normal `int` or `int32_t` made a difference, i dont notice much, so I guess it can be labeled just `int`?

thisiskeithb, 2023-05-25:
> looking for some feedback, comments, suggestions.

Please submit a PR for review.

> just wanted the ability to change the encoder rate to adjust it on the fly.

Why does this need to be a runtime feature?

Also: Don’t forget to keep other UIs, particularly Marlin UI, in mind as the standard/base UI to implement these types of features.

classicrocker883, 2023-05-26:
> a runtime feature

well not necessarily. I only wanted to be able to change the encoder rate without having to constantly recompiling and flashing firmware. I wasn't happy with the default, so I thought to go ahead and try to adjust it on the fly. 

this technically doesn't need to be a runtime thing but more so for debug in that sense.

im also trying to figure out capabilities of the programming language with my skills. my unpolished low skills.

I saw the new update for Mriscoc's ProUI and it has a few new features, which gave me a couple of ideas. I'm just not so familiar with how to go about coding.

so basically I'm looking for feedback as well, if the code I wrote is actually correct, or what would be changed to make it better. it does work how I intended but I'm sure it can be optimized.

as for a pull request I'll try to consider the other UI's if I can merge the code over simply.

thisiskeithb, 2023-05-26:
If your encoder is slow/you have a high-resolution encoder, you should enable `ENCODER_PULSES_PER_STEP` and increase the default value.