MarlinFirmware/Marlin #25883: [FR] Marlin firmware option or downloadable utility to completely erase the EEPROM

trengtor, 2023-05-26:
### Is your feature request related to a problem? Please describe.

Sometimes situations arise when it is necessary to completely reset (erase) an external EEPROM CHIP. 
I guess this is not too big a request to the developers: to provide such an opportunity, preferably without being tied to any version or branch of Marlin.
You can, of course, say that this can be done by soldering and erasing the EEPROM using a programmer, but such an action is always associated with a risk for the control board.

### Describe the feature you want

Marlin firmware option or downloadable utility to completely erase the EEPROM

EvilGremlin, 2023-05-26:
What is exact usecase? I can't imagine anything that require complete erase.

trengtor, 2023-05-26:
> What is exact usecase? I can't imagine anything that require complete erase.

For example: after rolling back from 2.1.2 to 2.0.8.2, it turned out that it was impossible to correctly initialize EEPROM.

ellensp, 2023-05-26:
M502 then M500 always resets eeprom back to defaults for the firmware 
As EEPROM_INIT_NOW  If the firmware has it

EvilGremlin, 2023-05-26:
I think some odd omission bug in some old version is not a valid reason to add new feature.

trengtor, 2023-05-26:
> M502 then M500 always resets eeprom back to defaults for the firmware As does EEPROM_INIT_NOW If the firmware has it

Not for this case

trengtor, 2023-05-26:
> And i doubt it'll help anyway. If 2.0.8.2 have no code to write some value tio EEPROM - full erase can't possibly fix that.

Думаю, что проблема здесь в том, как было изменено хранение (разметка и форматы) в ветке 2.1. И в том, что никто и никогда не думает об обратной совместимости. На твой вопрос, кстати, есть банальный ответ другим вопросом: а как, по-твоему, 2.0.8.2 накатывается на свежекупленные платы с девственно чистым EEPROM?

ellensp, 2023-05-26:
This is already in marlin
MARLIN_DEV_MODE
enables d-codes  D1 is Zero or pattern-fill the EEPROM data
```
    case 1: {
      // Zero or pattern-fill the EEPROM data
      #if ENABLED(EEPROM_SETTINGS)
        persistentStore.access_start();
        size_t total = persistentStore.capacity();
        int pos = 0;
        const uint8_t value = 0x0;
        while (total--) persistentStore.write_data(pos, &value, 1);
        persistentStore.access_finish();
      #else
        settings.reset();
        settings.save();
      #endif
      hal.reboot();

```

EvilGremlin, 2023-05-26:
It's really bad manners to switch language...
Formatting should be irrelevant. EEPROM write is complete write of all values from RAM, it's not "read, edit if found and write back"
And there was an issue where sometimes you had to reset 2-3 times to actually write everything.

ellensp, 2023-05-26:
And there is void GcodeSuite::M504()  in MARLIN_DEV_MODE
That lets you save any byte to any address in eeprom. 
So a simple gcode script can be used to erase eeprom this was too

trengtor, 2023-05-26:
> And there is void GcodeSuite::M504() in MARLIN_DEV_MODE That lets you save any byte to any address in eeprom. So a simple gcode script can be used to erase eeprom this way too

Thanks alot!

thisiskeithb, 2023-05-26:
> This is already in marlin
MARLIN_DEV_MODE
enables d-codes D1 is Zero or pattern-fill the EEPROM data

Closing since this already exists.

closed 2023-05-26 by thisiskeithb

trengtor, 2023-05-26:
Well, I made dev-version of 2.0.8.2 & tryed to use D1. 
But it useless: I see figures in printer statistics before D1 & I see the same figures after D1 (board making selfrebooot after that command).

trengtor, 2023-05-26:
> > This is already in marlin
> > MARLIN_DEV_MODE
> > enables d-codes D1 is Zero or pattern-fill the EEPROM data
> 
> Closing since this already exists.

You closed this issue too soon. I need some more help.

EvilGremlin, 2023-05-26:
Then it looks like generic issue with i2c eeprom handling in 2.0.8.2, there were some eeprom-related changes/improvenets around that time. Only thing you can do is to try cherypicking some later related commits i guess. Anyway, talking about what happened in old version is useless, this issue was fixed long time ago.

trengtor, 2023-05-26:
I would like to see examples of using D-codes: D1 and D3. These are not the most common methods.

thisiskeithb, 2023-05-26:
> I would like to see examples of using D-codes D3. These are not the most common methods.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

trengtor, 2023-05-27:
> This is already in marlin MARLIN_DEV_MODE enables d-codes D1 is Zero or pattern-fill the EEPROM data
>
> ```
>     case 1: {
>       // Zero or pattern-fill the EEPROM data
>       #if [...]

I tried to do it, but it doesnt working. 'It's so esy', yes. Bot no.

[image]

[image]

[image]

[image]

EvilGremlin, 2023-05-27:
Check if your chip is actually good, with a programmer. At least 10 cycles of write&verify on random data (new and unique each time), full chip size.

trengtor, 2023-05-27:
> Check if your chip is actually good, with a programmer. At least 10 cycles of write&verify on random data (new and unique each time), full chip size.

Come on, you'd better describe to me in detail the D-commands D1 and D3 with their syntax. But only with the syntax that will actually work. Or do you not see what Marlin said to the D1 team?

EvilGremlin, 2023-05-27:
"unknown syntax" message is from pronterface. Disable filtering (if it can do that) or use different terminal.

trengtor, 2023-05-27:
> "unknown syntax" message is from pronterface. Disable filtering (if it can do that) or use different terminal.

It's clear. But for some reason it doesn't work in another terminal either. Code D1 seems to be running, the board is rebooting. But it doesn't erase anything. But at the same time, saving the parameters works, statistics are also accumulated. What could be the problem? Maybe something in the method/object itself?

I repeat: it started after the rollback of the firmware from the 2.1 branch to the 2.0 branch.

EvilGremlin, 2023-05-27:
Could be. Could be partially dead chip that could behave in any bizarre manner.

>Well, I made dev-version of 2.0.8.2 & tryed to use D1.

Again, 2.0.8.2 wont receive any fixes, it's completely useless to talk about it's bugs. 2.0.9.6 might, so use at least that (for D-commands). Or 2.1.x.

trengtor, 2023-05-27:
> Could be. Could be partially dead chip that could behave in any bizarre manner.
> 
> > Well, I made dev-version of 2.0.8.2 & tryed to use D1.
> 
> Again, 2.0.8.2 wont receive any fixes, it's completely useless to talk about it's bugs. 2.0.9.6 might, so use at least that (for D-commands). Or 2.1.x.

With 2.0.9.6 all the same

trengtor, 2023-06-01:
Boards, supported in gcode_d.cpp (Marlin\src\gcode): 5 from 332 (1.5%).
I know that EEPROM is not available on all 332 boards, but in this context it does not matter.
Support for only 1.5% of the boards, that is, it is statistically negligible. In fact, we can say that this function (D1 and D3) is not in Marlin. And you still have this situation, including in bugfix 2.1. It is enough to look at the source code of this module to understand this.

[image]