MarlinFirmware/Marlin #25891: [BUG] D-codes don't work on modern boards

trengtor, 2023-05-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

D-codes don't work on modern boards.

Continuing the theme started in Issues 25151, 25788 and 25883

Let me remind the history of this problem.

After the firmware was rolled back (for the BTT E3 RRF V1.1 board) from the 2.1 branch (2.1.2 and bugfix-2.1 *) to the 2.0 branch (2.0.8.2), EEPROM initialization stopped working normally. This manifested itself in the form of incorrect values for the hybrid threshold parameters for the axes, because of which it was necessary to write them manually. Probably, the reason for this is the change in the formats (markup) for saving data in EEPROM, adopted in the 2.1 branch.

It became clear that the data in the EEPROM needed to be erased so that Marlin could fully initialize it, as happens on newly purchased boards. There was an idea to erase the EEPROM with a programmer by connecting it to the chip with a clip, but this turned out to be impossible (EEPROM packaged in TSSOP-8, not SOI / SOIC-8).

In 25883 it was suggested to me that the problem could be solved by enabling MARLIN_DEV_MODE and then using the D1 code.

Ok, I enabled MARLIN_DEV_MODE, and then the most interesting part of this story began. No matter what I did - EEPROM was not erased in any way. Each time after entering the D1 code, a short time passed, the board rebooted, but the data in the EEPROM was not erased. This was evident from the fact that after D1 code the previous data in the printer operation statistics did not change in any way. It was the same with the D3 code. I tested this in both 2.0.8.2 and 2.0.8.9. The maximum that active community members could advise in this situation was to check the physical performance of the EEPROM chip (known to be good, see above).

Well, since the programmers could not, I had to see for myself what the problem was. Here's what I was able to find out. In the gcode_d.cpp module, work with EEPROM is implemented through the eeprom_bl24cxx.cpp module. Which requires an EEPROM declaration via IIC_BL24CXX_EEPROM. It is important. The fact is that modern boards (such as the aforementioned BTT E3 RRF V1.1) use the definition of EEPROM as I2C_EEPROM with the simultaneous definition of MARLIN_EEPROM_SIZE.
Probably, there may be other options for determining EEPROM, but let your programmers check this already.

What is the result? gcode_d.cpp module does not perform its functions on modern boards. He simply does not see EEPROM on them. At the same time, everything is compiled and, as it were, executed.

I really hope that you will make the necessary changes to gcode_d.cpp not only in the 2.1 branch, but also in the 2.0 branch (2.0.9.6).

### Expected behavior

I really hope that you will make the necessary changes to gcode_d.cpp not only in the 2.1 branch, but also in the 2.0 branch (2.0.9.6).

### Actual behavior

gcode_d.cpp module does not perform its functions on modern boards. He simply does not see EEPROM on them. At the same time, everything is compiled and, as it were, executed.

### Version of Marlin Firmware

2.0.9.6

### Printer model

ZAV-mini

### Electronics

BTT E3 RRF V1.1

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/11583905/Marlin.zip)

thisiskeithb, 2023-05-28:
We’re not going to backport fixes to previously tagged versions.

Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

trengtor, 2023-05-28:
> We’re not going to backport fixes to previously tagged versions.
> 
> Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

2.0.9.6 is LTS version or no? I don't want to use 2.1.* because of the problems it has. The amendments in the 2.0.9.6 code, I believe, will not differ much from 2.1.*.

trengtor, 2023-05-28:
> We’re not going to backport fixes to previously tagged versions.
> 
> Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

In bugfix-2.1.x the same bugfix-2.1.x as in 2.0.9.6. Nothing changed.

thisiskeithb, 2023-05-28:
> I don't want to use 2.1.* because of the problems it has.

Just because you don’t want to use it, doesn’t mean this bug hasn’t been fixed.

`bugfix-2.1.x` includes EEPROM validation that was just added: https://github.com/MarlinFirmware/Marlin/commit/5bc84bf08cb47741537fe705dbfe19eee5396d4e

trengtor, 2023-05-28:
> > I don't want to use 2.1.* because of the problems it has.
> 
> Just because you don’t want to use it, doesn’t mean this bug hasn’t been fixed.
> 
> `bugfix-2.1.x` includes EEPROM validation that was just added: [5bc84bf](https://github.com/MarlinFirmware/Marlin/commit/5bc84bf08cb47741537fe705dbfe19eee5396d4e)

Since you are so categorical, then I reserve the right to express my opinion about the quality of support for the Marlin firmware and its code on any resources that I find suitable for this.
You released 2.0.9.6 in April this year. Is it really that hard to release 2.0.9.7 fixing this epic bug that got into dev mode?

thisiskeithb, 2023-05-28:
Again, please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

Does EEPROM Initialization (`M502` + `M500`) work for you in current bugfix?

trengtor, 2023-05-28:
I have nothing to add to what has been said. I need a printer for work, not for endless catching your bugs.

trengtor, 2023-05-28:
In 2.1 the same bug sits. But you can't read what isn't written by you, can you?

thisiskeithb, 2023-05-28:
Since you’re unwilling to test a current version of Marlin, I went ahead and did it.

I can confirm that EEPROM Initializing works fine.

Your EEPROM or something else on your board is likely damaged, so I’d recommend the following resources for additional help:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2023-05-28 by thisiskeithb

trengtor, 2023-05-28:
You are mudak

thisiskeithb, 2023-05-28:
Locking since you’ve resorted to name calling and are in clear violation of our [Code of Conduct](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/code_of_conduct.md).

thinkyhead, 2023-05-30:
Locked before I had any chance to chime in? Here are some of my thoughts for you @trengtor …

- The D codes are kind of experimental and were just added for developers to play around with. I haven't used them myself. That code might have issues, but they looked okay to me in my last visit to that code many months ago. There is some indication that Pronterface simply won't send those codes, but a basic serial terminal or other host should obediently send them on.
- Keith is well-meaning and intends to be helpful. There is no justification for you to be impatient and mean to him, make demands, and engage in rudeness. I agree with him that it makes sense to lock this issue and disengage. We do this work voluntarily for the joy of the process, and we have no obligation to deal with jerks.
- Keith (or whoever else happens to respond to an issue) does not represent the "support department" of Marlin. He is one contributor who helps with this project for fun in his spare time. You will get different responses from different contributors at different times. I would likely have responded differently from the start, because I like to find and fix problems, even with older versions of Marlin, if they are critical enough. You need to be more mature and kind while dealing with people, including those who make you feel offended. And, learn to take at least half of the responsibility when you feel offended.
- As I said, I am not personally averse to fixing issues in 2.0.x but my time is stretched very thin and I have other priorities. So you would need to be patient and not make entitled demands in any case. If you don't personally have the capability to fix an issue that you found, it is still not proper to demand that someone else in the world jump in and fix it. There simply isn't enough time and talent to address everything the moment that it gets reported.
- I will look into this and see what's going on, but no guarantees that it will be repaired very quickly, and we don't need to re-engage. You have made your report and demonstrated your disrespect. Thank you for the initial report, and I apologize that you did not get the response that you wanted.