MarlinFirmware/Marlin #25900: [FR] (Add new version Tramming wizard without hard coded positions)

nicwilson58, 2023-05-30:
### Is your feature request related to a problem? Please describe.

I have over 15,000 members in my group and I maintain several hundred free firmware.  There is always probe offset differences for many users,  and in ealier Marlin altering probe offsets could be done in the menu. Since the introduction of the Tramming Wizard with its hardcoded pre-calculated positions,  means if users alter the probe offsets in menu it breaks the tramming wizard, making the change probe offset menu redundant.
I understand at this late stage in its development and poplularity changing this behaviour would be difficult for firmware authors.  However leaving it as is,  and making a Tramming Wizard 2,  or an "auto position" option could be achieved where for example the four corners of the bed levelling could be used as reasonable corner positions for the tramming wizard.   Such a feature would save an enormous amount of time for those of us supporting thousands of users of firmware.

### Are you looking for hardware support?

N/A

### Describe the feature you want

Either Tramming Wizard 2 with no hardcoded positions,  or a settable define that causes the four corners of  bed level to be used as reasonable corner positons for tramming wizard,  or similar feature.

classicrocker883, 2023-05-30:
check out Mriscoc's ProUI, or here is my fork of it.

https://github.com/classicrocker883/MriscocProUI

what I did was do what I think youre suggesting.
this pertains to the same LCD type as Ender 3 V2, or Voxelab Aquila. I cant say what else this is compatible with, but it uses `DWIN_LCD_PROUI` which I'm sure can be ported to be used elsewhere.

anyway what I did was copy what JyersUI did, basically. So what youre able to do is set the `BED_SCREW_INSET`, this corresponds to the bed leveling knobs. for example, each of the 4 screws holes where the knob is used to tram the bed are roughly **1"** or **25.4mm** away from the sides of the bed.

when you are tramming/leveling the bed, manually or with a probe, it is best so that the nozzle or probe is directly above the center of the _knob_ / _screw_ / _spring_ which gives you a better and more accurate level. in order for the 3d printer to know where that is, you must set those values in the firmware.

what I did was make those values editable, such as if the screw inset is 25mm in from the X edge and Y edge, and the Home Offsets correctly place the nozzle/probe in the center of the bed, then it puts the tram positions equidistant and directly above the screw where it needs to be.

however, if you have a probe and the X (or Y) offset is too much, this will also effect the position where it can successfully tram. for instance one side will be fine, but the other will have to compensate for the offset of the probe. in this case, I think instead if all 4 points cant be above each bed leveling knob, then a good compromise would be setting each point equidistant on the bed. this will make the 4 corner "box" smaller, but it will be easier to adjust when leveling. again, if the travel distance is limited for the probe to reach the points, its better to have all the points the same - to create a Square with each point, rather than rectangle or oblong.

EvilGremlin, 2023-05-30:
If XY probe offset is calculated in at compile time - this is very much a bug. Only tramming points are intended to be hardcoded.