MarlinFirmware/Marlin #25906: Unable to compile for tronxy_stm32f446 and tronxy_stm32f103

phucly01, 2023-05-31:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I was just started on this project.  Downloaded the code and tried to compile it in VSCode/platformio 

It failed with the following output:

Executing task: C:\Users\Phuc\.platformio\penv\Scripts\platformio.exe run 

Warning! Ignore unknown configuration option `monitor_flags` in section [env]
Processing tronxy_stm32f446 (platform: ststm32@~12.1; board: tronxy_stm32f446zet; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/tronxy_stm32f446zet.html
PLATFORM: ST STM32 (12.1.1) > STM32F446ZE (128k RAM. 512k Flash)
HARDWARE: STM32F446ZET6 180MHz, 128KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 @ 4.10900.200819 (1.9.0)
 - framework-cmsis @ 2.50501.200527 (5.5.1)
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
Dependency Graph
|-- SPI @ 1.0
|-- EEPROM @ 2.0.1
|-- SoftwareSerial @ 1.0
|-- IWatchdog @ 1.0.0
|-- Servo @ 1.1.2
|-- Wire @ 1.0
Building in release mode
RuntimeError: deque mutated during iteration:
  File "C:\Users\Phuc\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 181:
    env.SConscript(env.GetExtraScripts("post"), exports="env")
  File "D:\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "D:\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "D:\Projects\Tronxy-Marlin-FW-main\buildroot\share\PlatformIO\scripts\tronxy_scrypt.py", line 8:
    marlin.relocate_vtab(board.get('build.offset'))
  File "D:\Projects\Tronxy-Marlin-FW-main\buildroot\share\PlatformIO\scripts\marlin.py", line 33:
    replace_define("VECT_TAB_OFFSET", address)
  File "D:\Projects\Tronxy-Marlin-FW-main\buildroot\share\PlatformIO\scripts\marlin.py", line 22:
    for define in env['CPPDEFINES']:
========================================================================================== [FAILED] Took 4.08 seconds ==========================================================================================

Environment       Status    Duration
----------------  --------  ------------
tronxy_stm32f446  FAILED    00:00:04.080
==================================================================================== 1 failed, 0 succeeded in 00:00:04.080 ==================================================================================== 

### Bug Timeline

Don't know, first time trying it.

### Steps to Reproduce

1. I followed the instruction on this site to download VSCode
2. Downloaded the source code
3. Opened it in VSCode and install platformio and other modules that prompted by VSCode.
4. Changed to tronxy_stm32f103 (tronxy_stm32f446 didn't compile either)
5. Click Build
6. And failed with error message above.

### Version of Marlin Firmware

Tronxy x5sa pro 1.32

### Printer model

Tronxy x5sa pro

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

There was no change, I was just tested to see if the source code compile.  
I compiled the source code as it and it failed.
Changed default_envs = tronxy_stm32f446 to default_envs = tronxy_stm32f103 and failed too.

closed 2023-05-31 by phucly01

EvilGremlin, 2023-05-31:
You use fork, not vanilla marlin, so you should address any issues to fork maintainer. 
This case looks like outdated code that new version of platformio don't understand.

phucly01, 2023-06-01:
Yes I realized I posted on the wrong board.  Too many tabs opened.  Thanks