MarlinFirmware/Marlin #25907: Enable Host Action Command support by default?

qwertychouskie, 2023-05-31:
Host Action Command support is required for Start/Stop/Pause/Resume functions on the printer to work when using Octoprint and similar software, and [Octoprint will show a warning if support is not enabled](https://community.octoprint.org/t/octoprint-tells-me-my-firmware-lacks-support-for-host-action-commands-what-does-this-mean/34588).  However, the default Marlin configs have Host Action Command support disabled by default.  Is there a reason for this?  I've enabled it even on old Creality 8-bit (1.1.x) boards with no issues.

Enabling it by default not only makes the experience better for users who are self-compiling, but also will significantly improve the out-of-the-box experience of new printers.  As manufacturers release new printers  based on newer versions of Marlin, they will have Host Action Command support out-of-the-box, whereas with the current situation, manufactures would have to manually enable it (and most don't).

I'd be happy to create a PR to enable it by default for all board (or alternatively if needed, all boards that meet some minimum spec), but wanted to get input before doing the work.

thisiskeithb, 2023-05-31:
The main issue is that it increases the default build size and while OctoPrint is commonly paired with Marlin, it’s not shipped as a standard option on most printers.

qwertychouskie, 2023-06-01:
> The main issue is that it increases the default build size and while OctoPrint is commonly paired with Marlin, it’s not shipped as a standard option on most printers.

In the main issue usage on the flash, or RAM usage?  If it's flash space, it should be reasonable to enable it by default on most boards, and only leave it disabled by default on boards that have a very small flash space.

qwertychouskie, 2023-06-01:
I just tested a build for the Ender 3 with a 1.1.4 (8 bit) board, the build without support is 126,024 bytes, and the build with support is 125,902 bytes (this is for an "unoptimized" build, an optimized build would likely be even smaller).  Also, reported SRAM use by the compiler was identical.  I don't think 122 bytes of flash will make a measurable difference for most machines, and if it does, there are other features it would make more sense to disable (perhaps arc support isn't super valuable for extreme-budget printers), though I would expect to hit that limit with very few, if any, boards.

ellensp, 2023-06-01:
Marlin default is for smallest build with the most common features.

In my view Host Action Command is not in this category at all.

I don't have it enabled on on of my 8 printers...  there is no need for it.

If your talking about this warning

```
#if NONE(HAS_RESUME_CONTINUE, HOST_PROMPT_SUPPORT)
  #warning "Your Configuration provides no method to acquire user feedback!"
#endif
```

where HAS_RESUME_CONTINUE is true on any of these

```
#if ANY(EXTENSIBLE_UI, IS_NEWPANEL, EMERGENCY_PARSER, HAS_ADC_BUTTONS, HAS_DWIN_E3V2)
  #define HAS_RESUME_CONTINUE 1

```
Then having any of these will satisfy it, without HOST_ACTION_COMMANDS and the sub-setting HOST_ACTION_COMMANDS

ellensp, 2023-06-01:
"the build without support is 126,024 bytes, and the build with support is 125,902 bytes"   this is a decrease with added features... clearly this is wrong.

qwertychouskie, 2023-06-01:
> "the build without support is 126,024 bytes, and the build with support is 125,902 bytes" this is a decrease with added features... clearly this is wrong.

I mistyped with/without, the point is it's a 122 byte difference.

> If your talking about this warning

I'm talking about the warning shown by Octoprint, seen in the link in my first post.

I get that not everyone uses Octoprint, but for those that do, having it work out-of-the-box is a big usability/UX improvement.  Other optional features are enabled by default, e.g. arc support, which takes around 3200b (over 25x more than HAC) and M503, which takes ~2700b (over 20x more than HAC).

Also, these are all just defaults.  They can always be changed on a need-by-need basis both in individual board configs (e.g. boards that have extra-extra-limited flash space) and by the user.

thomas-christiansen, 2024-04-29:
+1

I have the S1 Pro. I would really love to see that Host Action Commands get enabled by default, so pause, stop, change filament, filament runout, etc. works better when using the printer with Octoprint.