MarlinFirmware/Marlin #25909: [BUG] Marlin 2.1 skip steps - 2.0 stopped working

alx3dev, 2023-06-01:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Last install was Marlin 2.0.9.3 and everything worked fine. I didn't used my Ender3 some time, then I installed 2.1 and noticed skipping steps a lot. Also inline mode not work for me unless I add `Sx`, which also worked with 2.0.

Now I tried 2.0.9.3 version again, same as before and also downloaded new again, steppers won't move. I took all components from ender3 to test on another printer with DIY board, everything work as expected. I use default ender3 motors, but also tried with new 0.9 steppers, no difference.

I tried with ender3 default configuration, also with my own edition too, nothing changed. Also removed all folders inside `.platiformio/platforms` (linux system), nothing changed. Any help is more than appreciated.

### Bug Timeline

Started two weeks ago, since then I am trying to fix it

### Expected behavior

Expected to move steppers with `2.0` version, or without missing steps in `2.1`.

### Actual behavior

In `2.0` version motors not move at all, they started enabled, when I move them nothing happen, just became disabled.
In `2.1` version motors skip steps a lot, and that never happened (using lightweight laser with 10 mm/s speed).

### Version of Marlin Firmware

2.1.1

### Printer model

Creality Ender-3

### Electronics

board v4.2.2

### Bed Leveling

No Bed Leveling

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

alx3dev, 2023-06-01:
I don't how to include zip with config here, but same happen with default ender3 or edited.

ellensp, 2023-06-01:
This is what happens is you think you have a v222, but you really have a v427 motherboard the only difference is the step and direction io lines are swapped over, so the stepper will not move when use get the wrong one.

Physically verify you have a 4.2.2 motherbaord

alx3dev, 2023-06-04:
Does anyone have any idea why is not possible to use older firmware after 2.1 version?

closed 2023-06-08 by alx3dev