MarlinFirmware/Marlin #25918: Bug: Compile error Marlin\src\lcd\fontutils.h:68:20: error: redefinition of 'const char* get_utf8_,,,

RogerRepRap, 2023-06-02:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Loaded latest bugfix (zip) on 6/1/23, Made very minor changes to config, compiled and received error:
Marlin\src\lcd\fontutils.h:68:20: error: redefinition of 'const char* get_utf8_value_cb(const char*, read_byte_cb_t, lchar_t&)'
Bug report #25755 says  "This isn't possible since fontutils.cpp was renamed in https://github.com/MarlinFirmware/Marlin/commit/9a7d9e6995f4f66c4b48147ec7876d49d4b3ae77 and no longer exists."
[Marlin Config.zip](https://github.com/MarlinFirmware/Marlin/files/11638754/Marlin.Config.zip)
[platformio.zip](https://github.com/MarlinFirmware/Marlin/files/11638762/platformio.zip)

### Bug Timeline

New

### Version of Marlin Firmware

Latest bug fix as of 6/1

### Electronics

Octopus 1.1

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

EvilGremlin, 2023-06-02:
Imposible indeed, you downloaded something wrong (github glitch regularly) or pio grabbed old cache.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

* MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
* RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
* Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
* Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
* [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
* 3D Printing Discord server. Join link: https://discord.gg/3dprinters
* Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a bug report.

thisiskeithb, 2023-06-02:
> Marlin\src\lcd\fontutils.h:68:20: error: redefinition of 'const char* get_utf8_value_cb(const char*, read_byte_cb_t, lchar_t&)'

`fontutils.h` no longer exists in current `bugfix-2.1.x` and your config compiles fine once you update the recently dropped BOTH & EITHER macros.

```prolog
Linking .pio/build/STM32F446ZE_btt/firmware.elf
Checking size .pio/build/STM32F446ZE_btt/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   7.6% (used 10016 bytes from 131072 bytes)
Flash: [===       ]  27.9% (used 146056 bytes from 524288 bytes)
Building .pio/build/STM32F446ZE_btt/firmware.bin
============================================================= [SUCCESS] Took 55.38 seconds =============================================================

Environment      Status    Duration
---------------  --------  ------------
STM32F446ZE_btt  SUCCESS   00:00:55.380
============================================================== 1 succeeded in 00:00:55.380 ==============================================================
```

closed 2023-06-02 by thisiskeithb