MarlinFirmware/Marlin #25919: SKR 3 EZ (723) should have 1MB of flash

discip, 2023-06-02:
> Maybe I am missing here something, but according to `ST's` datasheet for the **`STM32H723VGT6`** it has **_1MB_** of flash. Why do we only use **_512KB_**?

We have multiple boards/MCUs sharing environments.

Please [open a bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose) so this can be tracked and I'll look into it.

_Originally posted by @thisiskeithb in https://github.com/MarlinFirmware/Marlin/issues/25387#issuecomment-1574361899_

The MCU I am seeing on my SKR 3 EZ states: STM32H723VGT6
And according to [ST's datasheet](https://www.st.com/en/microcontrollers-microprocessors/stm32h723vg.html) for this ship it should be equipped with 1MB of flash.

thisiskeithb, 2023-06-02:
Both the BigTreeTech SKR V3.0 / SKR V3.0 EZ (STM32H723VGT6) & Octopus Max EZ V1.0 (STM32H723VET6) share the same build environment, so it'll need to be split up since STM32H723VET6 only has 512K of flash.

discip, 2023-06-02:
Thank you for confirming this find. 👍🏻

ellensp, 2023-06-03:
related to this

When you attempt to build Octopus Max EZ V1.0 (STM32H723VET6)

You cannot as the It uses port F pins for endstops, which don't exist on the VE
Are we sure there is a STM32H723VE version? as the would need a new pins file.

Octopus Max EZ V1.0  (STM32H723ZET6)  builds as expected

ellensp, 2023-06-03:
@discip please test this PR

ellensp, 2023-06-03:
I'm confident this is fine, closing.

closed 2023-06-03 by ellensp

discip, 2023-06-03:
@ellensp
Thank you for fixing this!
Unfortunately I am still not able to test my configuration, as I am not able to figure out how to make the compilation work using the TMC2240 Steppers.
Maybe you have some advice for me on how to sync the repo I got from BigTreeTech with the latest bugfix? 🤷🏻

Here is the link with the provided repo: [Marlin-TMC2240-SPI-SKR3.rar](https://we.tl/t-vBdeYJk6Qs)
No clue how to proceed.
Already tried to rebase the provided repo, unfortunately without success. 😢

discip, 2023-06-03:
@ellensp
## Thank you for fixing this! 👍🏻

Unfortunately I still can not test my configuration, as I am not able to figure out how to make the compilation work using the TMC2240 Steppers.
Maybe you have some advice for me on how to sync the repo I got from BigTreeTech with the latest bugfix? 🤷🏻

No clue how to proceed.
Already tried to rebase the provided repo, unfortunately without success. 😢

Here is the link with the provided repo:
<details>
  <summary>
    <b>link</b>
  </summary>

[Marlin-TMC2240-SPI-SKR3.rar](https://we.tl/t-vBdeYJk6Qs)

</details>