MarlinFirmware/Marlin #25922: [BUG] Octopus Max EZ V1.0  wont build with env:STM32H723Vx_btt

ellensp, 2023-06-03:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

BTT_OCTOPUS_MAX_EZ_V1_0  lists these two env: STM32H723Vx_btt and STM32H743Vx_btt
STM32H743Vx_btt builds fine
But STM32H723Vx_btt will not

### Bug Timeline

Unknown

### Expected behavior

It should build

### Actual behavior

It errors with

```
Marlin/src/inc/../pins/stm32h7/pins_BTT_OCTOPUS_MAX_EZ.h:56:45: error: 'PF0' was not declared in this scope; did you mean 'PF_0'?
   56 | #define X_DIAG_PIN                          PF0   // M1-STOP

```
Port F pins are not defined in this varient

### Steps to Reproduce

1. #define MOTHERBOARD BOARD_BTT_OCTOPUS_MAX_EZ_V1_0
2. #define SERIAL_PORT -1
3. default_envs = STM32H723Vx_btt
4. build and watch it fail

### Version of Marlin Firmware

Bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-06-03:
Note after https://github.com/MarlinFirmware/Marlin/issues/25919 is committed env name will change to

from STM32H723Vx_btt to STM32H723VE_btt

thisiskeithb, 2023-06-03:
@bigtreetech: Does a second variant of this board exist?

sargonphin, 2023-06-04:
> Does a second variant of this board exist?

I have the STM32H723Zx variant, or so it says on the CPU at least. @ellensp do you have 2 options for the boards when you build?

[image]

thisiskeithb, 2023-06-04:
> I have the STM32H723Zx variant

Yes, that’s what I also have here: STM32H723ZET6.

thisiskeithb, 2023-06-04:
> BTT_OCTOPUS_MAX_EZ_V1_0 lists these two env: STM32H723Vx_btt and STM32H743Vx_btt
> STM32H743Vx_btt builds fine
> But STM32H723Vx_btt will not

I think you mean `STM32H723Zx_btt` & `STM32H723VE_btt` (no STM32H743-based version):

https://github.com/MarlinFirmware/Marlin/blob/f8423f9da09a8d67074b4243917953a6b2b8d279/Marlin/src/pins/pins.h#L816-L817

thisiskeithb, 2023-06-05:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/25934. The Octopus Max EZ only ships with the STM32H723ZET6 since it needs >100 pins for GPIO.

PR #25934 (2023-06-08): Remove STM32H723VET6-based Octopus Max EZ
### Description

I confirmed with BigTreeTech today that the Octopus Max EZ only ships with an STM32H723ZET6 MCU since it needs >100 pins for GPIO.

Resolves #25922

### Related Issues

- #25922
- #25921
closed 2023-06-05 by thisiskeithb