MarlinFirmware/Marlin #25926: Unable to compile chitu_f103_maple code using the documented configuration.

phucly01, 2023-06-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Ok so this is my second time here and I messed up the first time so please bear with me.  Just for a bit of a background.  I attempted to compile a baseline for my tronxy x5sa pro so I have a working code to work with.  Ultimately I am trying to add code to compensate the inaccuracy of the prints.  So far I already tried two repositories that generate firmware for the printer I have, supposedly.  Yet, they both didn't compile.  So My attempt is to use this code to make the firmware.  
I followed the instruction here: https://github.com/MarlinFirmware/Configurations/blob/release-2.0.9.3/config/examples/Tronxy/X5SA/HOWTO-INSTALL.md to set the configuration.  How ever it yield the compilation error.  
I don't think it is the error from the code.  I think it is more likely the platformio issue.  I hope some senior members can point me to the right direction.

Additional info:
VSCode: 1.87.2
Platformio: 6.1.7

This is the error:
Processing chitu_f103_maple (board: marlin_maple_CHITU_F103; platform: ststm32@~12.1; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/marlin_maple_CHITU_F103.html
PLATFORM: ST STM32 (12.1.1) > CHITU STM32F103Z (64k RAM. 512k Flash)
HARDWARE: STM32F103ZET6 72MHz, 59.12KB RAM, 469.03KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32-maple @ 3.10000.201129 (1.0.0)
 - tool-stm32duino @ 1.0.2
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- SoftwareSerialM @ 1.0.0
|-- STM32ADC @ 1.0
|-- EEPROM
|-- Wire @ 1.0
|-- USBComposite for STM32F1 @ 0.99
|-- Servo(STM32F1) @ 1.1.2
Building in release mode
Compiling .pio\build\chitu_f103_maple\src\src\HAL\STM32F1\HAL.cpp.o
Compiling .pio\build\chitu_f103_maple\src\src\HAL\STM32F1\HAL_SPI.cpp.o
Compiling .pio\build\chitu_f103_maple\src\src\HAL\STM32F1\MarlinSerial.cpp.o
Compiling .pio\build\chitu_f103_maple\src\src\HAL\STM32F1\SPI.cpp.o
Compiling .pio\build\chitu_f103_maple\src\src\HAL\STM32F1\Servo.cpp.o
Compiling .pio\build\chitu_f103_maple\src\src\HAL\STM32F1\dogm\u8g_com_stm32duino_swspi.cpp.o
Compiling .pio\build\chitu_f103_maple\src\src\HAL\STM32F1\eeprom_bl24cxx.cpp.o
Compiling .pio\build\chitu_f103_maple\src\src\HAL\STM32F1\eeprom_flash.cpp.o
Compiling .pio\build\chitu_f103_maple\src\src\HAL\STM32F1\eeprom_if_iic.cpp.o
Compiling .pio\build\chitu_f103_maple\src\src\HAL\STM32F1\eeprom_sdcard.cpp.o
Compiling .pio\build\chitu_f103_maple\src\src\HAL\STM32F1\eeprom_wired.cpp.o
Compiling .pio\build\chitu_f103_maple\src\src\HAL\STM32F1\fast_pwm.cpp.o
In file included from Marlin\src\HAL\STM32F1\../../inc/../HAL/HAL.h:30:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:33,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/../HAL/../HAL/STM32F1/HAL.h:172:1: error: expected ';' before 'typedef'
 typedef int8_t pin_t;
 ^~~~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/../HAL/HAL.h:30:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:33,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/../HAL/../HAL/STM32F1/HAL.h:97:3: error: static assertion failed: SERIAL_PORT must be from 1 to 5. You can also use -1 if the board supports Native USB.
   static_assert(false, "SERIAL_PORT must be from 1 to " STRINGIFY(NUM_UARTS) ". You can also use -1 if the board supports Native USB.")
   ^~~~~~~~~~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/../HAL/HAL.h:30:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:33,
                 from Marlin\src\HAL\STM32F1\MarlinSerial.cpp:25:
Marlin\src\HAL\STM32F1\../../inc/../HAL/../HAL/STM32F1/HAL.h:172:1: error: expected ';' before 'typedef'
 typedef int8_t pin_t;
 ^~~~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/../HAL/HAL.h:30:0,
[... 176 lines cut ...]
Tronxy x5sa pro V1.32

### Printer model

Tronxy x5sa pro

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11644632/Configuration.zip)

ellensp, 2023-06-04:
That read me is not a full list of changes, just the highlightst

Look at the example config provided

#define SERIAL_PORT 1
#define SERIAL_PORT_2 3

GMagician, 2023-06-04:
OT: I see that STRINGIFY is not working in assert

ellensp, 2023-06-04:
@GMagician it does
but the assert is printers as well as the error line

Marlin\src\HAL\STM32F1../../inc/../HAL/../HAL/STM32F1/HAL.h:97:3: error: static assertion failed: SERIAL_PORT must be from 1 to 5. You can also use -1 if the board supports Native USB.

and 

static_assert(false, "SERIAL_PORT must be from 1 to " STRINGIFY(NUM_UARTS) ". You can also use -1 if the board supports Native USB.")

GMagician, 2023-06-04:
ops missed first!

phucly01, 2023-06-04:
Thank you Ellensp and GMagician
I got it compiled now but the printer doesn't recognize the cbd. I am trying to figure out how to set the platformio to keep the bin file to try it out.

phucly01, 2023-06-04:
> Thank you Ellensp and GMagician I got it compiled now but the printer doesn't recognize the cbd. I am trying to figure out how to set the platformio to keep the bin file to try it out. It looks like platformio removed the bin file after encrypting it.

I tried the bin file and it still didn't update.  I am trying to update the tronxy's official firmware to this firmware.  Is it doable or I have to do something in between like some sort of conversion?

phucly01, 2023-06-04:
I managed to put the bin file in using the STM32CubeProgrammer.  However, the screen is black.  Am I missing something in the configuration?

EvilGremlin, 2023-06-04:
Both boot and marlin, with correct addresses?

phucly01, 2023-06-04:
I am not sure what you mean by boot and marlin.  
I compiled the code and flash it with STM32CubeProgrammer at 0x08000000

EvilGremlin, 2023-06-04:
This won't work, env expect bootloader `board_build.address  = 0x0800A000` you have to chabge that to 0x08000000

phucly01, 2023-06-04:
Can you elaborate this a bit.  I just started this two days ago.
I am using the chitu_f103 env and made some changes in the configuration.h to meet the tronxy x5sa pro settings.
Am I supposed to make the bootloader and the firmware.bin and flash them both?

EvilGremlin, 2023-06-04:
However you like really, but first build and flash marlin without offset as you don't have bootloader anymore anyway

EvilGremlin, 2023-06-04:
Ok, i'm not so blind in the end... it's a mess, `chitu_f103_maple` env expect bootloader (thus encrypt bin) but missing `board_build.offset`
Delete `buildroot/share/PlatformIO/scripts/chitu_crypt.py` fron env and it should work

phucly01, 2023-06-04:
chitu_f103_maple didn't compile.  I have been using chitu_f103.  I also did what you recommended but the screen does not turn on so I have no idea if the firmware even run.  The download seems to gone through fine.

EvilGremlin, 2023-06-04:
Download is always fine, it will flash absolutely whatever, there are no checks.
Yep, boot is gone because you flashed with no offset. You'll have to find some opensource boot, configure and build for this board (not easy). Or find somewhere dump of stock one, which i cant google up. Eh, good riddance i say :)
How you env look like now?

phucly01, 2023-06-04:
How do you put the offset?  in the programmer or in the compilation?  
In STM32CubeProgrammer, each time I open the bin file and set to 0x08000000 it says address out of range.  There is no option to tell it to start at 0x08000000

EvilGremlin, 2023-06-04:
Bboth or neither i guess? 0x08000000 is flash start, no offset.
I don't use cube though, so no idea if it checks anything

EvilGremlin, 2023-06-04:
this is unreadable, use code brackets ( triple ```  )

EvilGremlin, 2023-06-04:
looks right, should work if it can at all

phucly01, 2023-06-04:
So I think the problem is the programmer didn't write to the correct offset.
Will setting this line force the offset?
board_build.offset   = 0x08000000
What program do you use that can tell the it the offset?

EvilGremlin, 2023-06-04:
Nope, without bootloader you don't set any offsets anywhere. just open bin and flash

phucly01, 2023-06-04:
so the bin file must have the bootloader prepended?

EvilGremlin, 2023-06-04:
no

phucly01, 2023-06-04:
I have the bootloader.  What do I do to get this to work?

EvilGremlin, 2023-06-04:
Forget about bootloader for now, just build like this and flash, don't touch any address settings in cube.
```
[env:chitu_f103]
extends              = stm32_variant
board                = genericSTM32F103ZE
#board_build.crypt_chitu = update.cbd
board_build.variant  = MARLIN_F103Zx
#board_build.offset   = 0x8800
build_flags          = ${stm32_variant.build_flags}
                       -DSTM32F1xx
build_unflags        = ${stm32_variant.build_unflags}
#extra_scripts        = ${stm32_variant.extra_scripts}
#                       buildroot/share/PlatformIO/scripts/chitu_crypt.py
```

phucly01, 2023-06-04:
I did that, it didn't work

EvilGremlin, 2023-06-04:
Then it doesn't work at all now, so flash back boot and then flash stock firmware from SD

phucly01, 2023-06-04:
Yup I did that, back to ground 0 now.
So, if I understand your correctly, this code set isn't working for my printer.

EvilGremlin, 2023-06-04:
Seems like it.
Of course you might be omitting some crucial thing that breaks everything as is often the case, only so much can be guessed remotely

phucly01, 2023-06-04:
Yeah,maybe I missed something but I won't know.  I did everything from the document for the my printer.

closed 2023-08-15 by github-actions[bot]