MarlinFirmware/Marlin #25933: [FR] Systematic G34 "Z Steppers Auto-Alignment" right before G28 "Auto home"

sargonphin, 2023-06-04:
### Is your feature request related to a problem? Please describe.

I just finished building a brand-new firmware for my Tronxy CoreXY and I am running into a calibration problem.

The previous setup was using the stock hardware configuration of the board for the Z steppers, namely it had both stepper motors connected to the only Z stepper driver (it's wired like that by default). They were "synchronized" with a loose belt just to keep them from desynchronizing when the steppers are off.

I have upgraded it to dual Z stepper driver and removed the belt, but now there is a strong possibility that when the steppers are off the bed comes out of alignment. I put the belt back to at-least limit the desynchronization.

### Are you looking for hardware support?

I am running a Tronxy CoreXY with a BTT SKR V1.4 Turbo board with TMC2209 stepper drivers (and a BTT TFT35 V3.0 but I don't thinkif it's relevant). **The bed probe is inductive and is configured as Z-MIN and Z-PROBE.**

Pulled latest bugfix-2.1.x branch firmware a few hours ago and started the config from scratch

### Describe the feature you want

Add a firmware option to enable Z steppers auto-alignment before any Z homing command to ensure that the bed is always level before doing anything on the printer

rondlh, 2023-06-05:
I think this will be impossible, because you first need to know where the Z-probe is before you can do Z-probing.
What I do is this... 
1. Home the printer (G28)
2. Do an auto Z-alignment (G34), it will do a homing first if you didn't do the G28 in step 1. I also enable HOME_AFTER_G34 (configuration_adv.h)
3. Do a bed leveling (G29)

That should do the trick.

sargonphin, 2023-06-05:
Of  course it is possible, altering the homing sequence to implement G34 on any Z homing would do the trick :) A simple option like G34_ON_Z_HOMING would work. Knowing the position of the Z axis is not even needed, since you are triggering the Z_PROBE in the process anyway

sargonphin, 2023-06-05:
For now the workaround is to change the start GCODE in our slicer to use G34 instead of G28 (the G34 command has homing steps in its function) but it would be nicer if *all* Z-homing could immediately contain Z-stepper auto-alignment. So more or less the process that @rondlh described, minus the bed leveling (not needed as the bed is quite stiff and doesn't change in shape that easily)

thisiskeithb, 2023-06-05:
There is some "cart before the horse" issues here. X/Y/Z positions must be known before `G34` will start.

> For now the workaround is to change the start GCODE in our slicer to use G34 instead of G28 

Since homing is required (unless X/Y/Z position is known) for both `MECHANICAL_GANTRY_CALIBRATION` & `Z_STEPPER_AUTO_ALIGN` variants of `G34`, that will work and is recommended for your use case.

What you're really after are macros that will change default g-code behaviors, which will be added in:

- https://github.com/MarlinFirmware/Marlin/pull/16209

closed 2023-06-05 by thisiskeithb