MarlinFirmware/Marlin #25936: [BUG] (TMC2240 not supported)

discip, 2023-06-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Trying to compile the latest bugfix-2.1.x using TMC2240 stepper-drivers always results in an error.
I was provided with the following repo by a seller on Ali... set up to support TMC2240.
<details><summary><b>link</b></summary>

[Marlin-TMC2240-SPI-SKR3.rar](https://we.tl/t-vBdeYJk6Qs)</details>
Unfortunately, this is an outdated repo and I haven't been able to locate the relevant changes to merge them into the current bugfix.

### Bug Timeline

latest bugfix-2.1.x

### Expected behavior

Marlin should compile with TMC2240 stepper-drivers.

### Actual behavior

Compilation ends in an error.

### Version of Marlin Firmware

https://github.com/MarlinFirmware/Marlin/commit/70288c6c4fae6a6334d7c48b9ef17da7ecf3c173

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[configuration-files.zip](https://github.com/MarlinFirmware/Marlin/files/11654148/configuration-files.zip)

thisiskeithb, 2023-06-05:
This is not a bug since they are not supported here (yet). Upstream support in the main teemuatlut/TMCStepper repo is pending:

- https://github.com/teemuatlut/TMCStepper/pull/270

thisiskeithb, 2023-06-05:
You can leave the feature request open, but just know that we are working on it.

thisiskeithb, 2024-04-16:
Closing since we have an open PR:

- https://github.com/MarlinFirmware/Marlin/pull/25974

closed 2024-04-16 by thisiskeithb