MarlinFirmware/Marlin #25941: Recv: Error:Thermal Runaway, system stopped! Heater_ID: E0[BUG] (bug summary)

Glod76, 2023-06-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Recv: Error:Thermal Runaway, system stopped! Heater_ID: E0

Printer rebooted

### Bug Timeline

newest firmware

### Actual behavior

Recv: Error:Thermal Runaway, system stopped! Heater_ID: E0

new changes 
switched from 
#define PIDTEMP         // See the PID Tuning Guide at https://reprap.org/wiki/PID_Tuning
//#define MPCTEMP           // ** EXPERIMENTAL ** See https://marlinfw.org/docs/features/model_predictive_control.html

to

//#define PIDTEMP         // See the PID Tuning Guide at https://reprap.org/wiki/PID_Tuning
#define MPCTEMP           // ** EXPERIMENTAL ** See https://marlinfw.org/docs/features/model_predictive_control.html

### Version of Marlin Firmware

Recv: FIRMWARE_NAME:Marlin bugfix-2.1.x (Jun  6 2023 21:17:25) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Donnie's FT-5 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff Recv: Cap:SERIAL_XON_XOFF:0 Recv: Cap:BINARY_FILE_TRANSFER:0 Recv: Cap:EEPROM:1 Recv: Cap:VOLUMETRIC:1 Recv: Cap:AUTOREPORT_POS:0 Recv: Cap:AUTOREPORT_TEMP:1 Recv: Cap:PROGRESS:0 Recv: Cap:PRINT_JOB:1 Recv: Cap:AUTOLEVEL:0 Recv: Cap:RUNOUT:0 Recv: Cap:Z_PROBE:0 Recv: Cap:LEVELING_DATA:1 Recv: Cap:BUILD_PERCENT:0 Recv: Cap:SOFTWARE_POWER:0 Recv: Cap:TOGGLE_LIGHTS:0 Recv: Cap:CASE_LIGHT_BRIGHTNESS:0 Recv: Cap:EMERGENCY_PARSER:1 Recv: Cap:HOST_ACTION_COMMANDS:1 Recv: Cap:PROMPT_SUPPORT:1 Recv: Cap:SDCARD:1 Recv: Cap:MULTI_VOLUME:0 Recv: Cap:REPEAT:0 Recv: Cap:SD_WRITE:1 Recv: Cap:AUTOREPORT_SD_STATUS:0 Recv: Cap:LONG_FILENAME:0 Recv: Cap:LFN_WRITE:0 Recv: Cap:CUSTOM_FIRMWARE_UPLOAD:0 Recv: Cap:EXTENDED_M20:0 Recv: Cap:THERMAL_PROTECTION:1 Recv: Cap:MOTION_MODES:1 Recv: Cap:ARCS:1 Recv: Cap:BABYSTEPPING:1 Recv: Cap:EP_BABYSTEP:0 Recv: Cap:CHAMBER_TEMPERATURE:0 Recv: Cap:COOLER_TEMPERATURE:0 Recv: Cap:MEATPACK:0 Recv: Cap:CONFIG_EXPORT:1 Recv: area:{full:{min:{x:-6.00,y:-28.00,z:0.00},max:{x:300.00,y:300.00,z:380.00}},work:{min:{x:0.00,y:0.00,z:0.00},max:{x:300.00,y:300.00,z:380.00}}}

### Printer model

Folgertech FT5 R2 Modded with Skr 1.4turbo

### Electronics

Folgertech FT5 R2 Modded with Skr 1.4turbo

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Other (explain below)

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/11671965/Configs.zip)

ellensp, 2023-06-07:
You need to tune MPC to your hotend and heater https://marlinfw.org/docs/gcode/M306.html  T parameter

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

ellensp, 2023-06-07:
also https://marlinfw.org/docs/features/model_predictive_control.html

closed 2023-06-07 by ellensp