MarlinFirmware/Marlin #25949: More than 1 Filament sensor does not compile on SKR v1.3

anger2016, 2023-06-08:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin 2.1.2.1 latest and also earlier does not compile with more than 1 filament sensor (motion type and normal type). 2 Filament_Runout_pins are defined (eg P1_28 and P1_26 in SKR 1.3 pins file. Error after compilation states M600 must be used with T%c. Advanced_pause and Nozzle_park features are enabled. Single Filament_Runout_Pin works ok.
Does someone know how to get it to work? All help is appreciated.
Thank you so much.

### Bug Timeline

Never got it to work on any Marlin version.

### Expected behavior

Should compile when pins are defined and related features are defined.

### Actual behavior

Works on single Filament_Runout_pin, but not with 2 Filament_Runout_pins. Cannot place 2 Filament runout pins in a row on single pin as 2 different extruders are used. 

### Steps to Reproduce

see above

### Version of Marlin Firmware

Marlin 2.1.2.1 and Marlin bugfix 2.1.X

### Printer model

Anet A8 modified

### Electronics

SKR V1.3 with TMC 2209 V1.2

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Same as my slicer

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

ERROR: FILAMENT_RUNOUT_SCRIPT needs a %c parameter (e.g., "M600 T%c") when NUM_RUNOUT_SENSORS is > 1

thisiskeithb, 2023-06-08:
> ERROR: FILAMENT_RUNOUT_SCRIPT needs a %c parameter (e.g., "M600 T%c") when NUM_RUNOUT_SENSORS is > 1

Update `FILAMENT_RUNOUT_SCRIPT` as the error suggests.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2023-06-08 by thisiskeithb

anger2016, 2023-06-09:
Dear Keith Bennet,Thank you for your email. However, since this issue of several filament sensors not compiling on Visual Studio Code on Skr V1.3 seems not to be solved, allow me to suggest to keep it open. Thank you very much for your assistance.Best regards, A.G.

Closed #25949 as completed.

ellensp, 2023-06-09:
@anger2016 please re read the reply and the error message.

Issue is a configuration issue

you have #define FILAMENT_RUNOUT_SCRIPT probably set to the default "M600"     (cannot say for sure as you didn't attach configuration files as was requested)  

when with two or more runout sensors you need 

#define FILAMENT_RUNOUT_SCRIPT "M600 T%c"

As the error says, your FILAMENT_RUNOUT_SCRIPT is missing the required %c

This is a user errer, not a marlin error. Fixing users is beyond our abilities.

anger2016, 2023-06-09:
Dear ellensp,

 

thank you very much for your reply. Tried your suggestion, and now it is working.

Allow me to thank you very much indeed for your advice and support. 

Appreciating your help. 

Have a nice day.

 

Best Regards

A.G.

 

Gesendet: Freitag, 09. Juni 2023 um 14:23 Uhr