MarlinFirmware/Marlin #25951: [BUG] TwoTrees BLU-5 Bluer Plus Marlin bugfix 2.1.x 230608 Display Issues

helakejr, 2023-06-08:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin displays do not function for TwoTrees BLU-5 Bluser Plus MKS_ROBIN_TFT43 TOUCH_PORTRAIT display.

NOTE: 
- Reference [BUG] TwoTrees BLU-5 Bluer Plus Marlin Menu Problems #25475.
- The MKS_ROBIN_TFT43 is a horizontal display, however when touch screen calibrated TOUCH_ORIENTATION TOUCH_PORTRAIT
- MKS_ROBIN_TFT43 has no physical controls and relies on the last line of menu as PAGE_UP,PAGE_DOWN and BACK

Display Issues:
Top Level Screen for 480x272 resolution:
- Doesn't fit on display in the vertical
- Nozzle Temp, Bed Temp, Fan Speed Control Menus not enabled
[image]
[image]

E, X, Y, Z Offset Screen for 480x272 resolution:
- Doesn't fit on screen
- Has been reformatted incorrectly

[image]

Too many lines displayed in all menus

[image]
[image]

### Bug Timeline

Marlin bugfix 2.1.x 230608

### Expected behavior

Top Level Screen for 480x272 resolution:
- Should fit on display in the vertical and not be clipped
- Nozzle Temp, Bed Temp, Fan Speed Control Menus enabled
[image]

E, X, Y, Z Offset Screen for 480x272 resolution:
- Should fit on screen
- Screen formatted correctly

[image]

One fewer lines displayed in all menus with last line reserved for control PAGE_UP,PAGE_DOWN and BACK

[image]

### Actual behavior

Top Level Screen for 480x272 resolution: Doesn't fit on display in the vertical, Nozzle Temp, Bed Temp, Fan Speed Control Menus not enabled

E, X, Y, Z Offset Screen for 480x272 resolution: Doesn't fit on screen, Has been reformatted incorrectly

Too many lines are displayed in all menus over-writing last line (control line) with a menu item.

### Steps to Reproduce

Marlin 

From Top Level screen 
1. verify that last line is partially clipped
2. Verify that selecting Nozzle Temp, Bed Temp, and Fan Speed
3. Verify that no menu is displayed
4. Select Z offset
5. Verify that E, X, Y, Z Screen is displayed
6. Verify that the display does not fit on screen and formatting unlike before:
[image]
7. Select any menu option from the main menu
8. Verify the screen, displays on too many lines overwriting Option ICONS on the last line of display.
9. Verify ICONs are inoperable. Pressing any ICON selects last line of menu display.
10. Verify only way to return to prior menu is to select the 1st line.

### Version of Marlin Firmware

Marlin bugfix 2.1.x 230608

### Printer model

Two Trees BLU-5 Bluer Plus

### Electronics

Stock MKS_Robin_Nano_v2, MKS_Robin_TFT43

### Add-ons

N/A

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Slic3r

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/11694078/Marlin.zip)

Menu Issues due to Marlin\src\inc\Conditionals_LCD.h code:
IS:
#if ANY(HAS_UI_320x240, HAS_UI_480x320, HAS_UI_480x272)
  #if ENABLED(TFT_COLOR_UI_PORTRAIT)
    #define LCD_HEIGHT TERN(TOUCH_SCREEN, 8, 9) // Fewer lines with touch buttons onscreen
  #else
    #define LCD_HEIGHT TERN(TOUCH_SCREEN, 6, 7) // Fewer lines with touch buttons onscreen
  #endif

WAS:
#if ANY(HAS_UI_320x240, HAS_UI_480x320, HAS_UI_480x272)
  #define LCD_HEIGHT TERN(TOUCH_SCREEN, 6, 7)   // Fewer lines with touch buttons onscreen

SUGGEST:
#if ANY(HAS_UI_320x240, HAS_UI_480x320, HAS_UI_480x272)
  #if (TFT_COLOR_UI_PORTRAIT && !HAS_UI_480x272)
    #define LCD_HEIGHT TERN(TOUCH_SCREEN, 8, 9) // Fewer lines with touch buttons onscreen
  #else
    #define LCD_HEIGHT TERN(TOUCH_SCREEN, 6, 7) // Fewer lines with touch buttons onscreen
  #endif

Other display issues dur to heavy modifications between Bugfix 230306 in Marlin\src\lcd\tft\ui_480x320.cpp and ui_480x320.h

helakejr, 2023-06-08:
Expected Top Level Menu should be:
[image]

uncellon, 2023-06-11:
I checked your config and came to the conclusion that for some reason `TFT_DEFAULT_ORIENTATION` for `MKS_ROBIN_TFT43` is 0. So a 0 or 180 degree rotation causes the `TFT_COLOR_UI_PORTRAIT` flag to be set to 1. And what you're seeing is an interface for portrait orientation. I don't know how exactly to interpret the `TFT_DEFAULT_ORIENTATION` flag and I don't know the characteristics of MKS_ROBIN_TFT43. But in my opinion, the snag is setting the value of `TFT_DEFAULT_ORIENTATION` for this display.

uncellon, 2023-06-11:
And yes, indirectly adding support for vertical orientation can be called a problem. But in fact, it's all about setting the `TFT_COLOR_UI_PORTRAIT` flag, which for some reason, as I said earlier, is 1 in your case. Maybe you can manually change the value of `TFT_DEFAULT_ORIENTATION` and test the changes?

helakejr, 2023-06-11:
I am setting because this is what the results of the M995 calibration
produces.
Note: when TOUCH_LANDSCAPE is used the touchscreen will not calibrate.

 #define TOUCH_CALIBRATION_X 16541
  #define TOUCH_CALIBRATION_Y 10702
  #define TOUCH_OFFSET_X        -17
  #define TOUCH_OFFSET_Y        -30
  #define TOUCH_ORIENTATION TOUCH_PORTRAIT

uncellon, 2023-06-11:
As far as I understand `TOUCH_PORTRAIT` and `TFT_COLOR_UI_PORTRAIT` are fundamentally different things. I'm not saying anything against your touch settings. I see a problem with displaying a portrait interface instead of landscape on your device and therefore wrote about `TFT_DEFAULT_ORIENTATION` value. And so I asked, if possible, to check the behavior when changing the value of `TFT_DEFAULT_ORIENTATION` in `Conditionals_LCD.h` file.

helakejr, 2023-06-11:
No changes from prior builds.

I will 1st try:
TFT_DEFAULT_ORIENTATION TFT_NO_ROTATION

helakejr, 2023-06-11:
TFT_DEFAULT_ORIENTATION TFT_NO_ROTATION produced exactly the same results as TFT_NO_ROTATION is zero.

helakejr, 2023-06-11:
I just noticed that in Configuration.h: #define TFT_ROTATION TFT_ROTATE_180. Could this be why the flag is getting set?

jmz52, 2023-08-11:
Color UI layout was reworked in https://github.com/MarlinFirmware/Marlin/pull/26077
Please retest using latest bugfix-2.1.x code

closed 2023-10-22 by github-actions[bot]