MarlinFirmware/Marlin #25953: [BUG] in temperature.cpp

mriscoc, 2023-06-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Error in the definition of _MINMAX_TEST(N,M)

https://github.com/MarlinFirmware/Marlin/blob/f3cda01b959274e7930c2050f5e34aa1460791bc/Marlin/src/module/temperature.cpp#LL2946C6-L2946C6

### Bug Timeline

Current bugfix

### Expected behavior

_MINMAX_TEST must be defined as:

```cpp
    #define _MINMAX_TEST(N,M) (HOTENDS > N && TEMP_SENSOR_##N##_IS_THERMISTOR && defined(HEATER_##N##_##M##TEMP))
```

### Actual behavior

_MINMAX_TEST is defined as:

```cpp
    #define _MINMAX_TEST(N,M) (HOTENDS > N && TEMP_SENSOR_##NR##_IS_THERMISTOR && defined(HEATER_##N##_##M##TEMP))
```

### Version of Marlin Firmware

Latest Marlin Bugfix

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

mriscoc, 2023-06-09:
PR: https://github.com/MarlinFirmware/Marlin/pull/25954

thisiskeithb, 2023-06-09:
FYI: To save some time/work for yourself, you don't need to create an issue if you plan to submit a PR with a fix.