MarlinFirmware/Marlin #25960: [BUG] Simpler distinct firmware rename omits *.bin*

classicrocker883, 2023-06-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

the terminal outputs the file like this - without the exact label suffix (date/time)
```
Building .pio\build\STM32F103RE_creality\firmware.bin
rename_target([".pio\build\STM32F103RE_creality\firmware.bin"], [".pio\build\STM32F103RE_creality\firmware. Elf"])
```

it used to print out like this for example
```
Checking size .pio\build\STM32F103RC_creality_maple\firmware-20230610-005630.elf
Building .pio\build\STM32F103RC_creality_maple\firmware-20230610-005630.bin
```

and when you go see the file in the folder destination - it does not have*`.bin`*, the file name is correct, but there is no type, it is blank.

### Expected behavior

it should be like the previous printout

### Actual behavior

does not show firmware-(*date*/*time*).bin
and does not have *.bin* file type in folder.

### Steps to Reproduce

the following `default_envs` affected with the recent change

**env:**

1. STM32F401RC_creality
2.  STM32F103Rx_creality
3. STM32F103RE_creality_maple
4. STM32F103RC_creality_maple
5. GD32F303RE_creality_maple
6. GD32F303RC_creality_maple
7. GD32F103RET6_sovol_maple

but may not be limited to these.

Steps To Reproduce:

1. platformio.ini `default_envs =` STM32F103RE_creality or *RC*
2. `BOARD_CREALITY_V4` or similar
3. Build

### Version of Marlin Firmware

Most recent commit firmware rename

### Printer model

Voxelab Aquila, ender-3 v2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

from the **/ini/stm32f*.ini** file -
in this code `board_build.rename   = firmware-{date}-{time}` 

i simply put *.bin* at the end like

```
board_build.rename   = firmware-{date}-{time}.bin
```

and it solved the problem of it not having a *.bin* type in destination folder, however the other issue of the terminal not printing out the true file name remains.

classicrocker883, 2023-06-10:
in the very last line of **buildroot/share/Platformio/scripts/marlin.py**
```
    env.AddPostAction(str(Path("$BUILD_DIR", "${PROGNAME}.bin")), action);
```
is that ` ; ` supposed to be there?
its showing as an error for me.

classicrocker883, 2023-06-10:
in **buildroot/share/Platformio/scripts/offset_and_rename.py**

```
        def rename_target(source, target, env):
            from pathlib import Path
            from datetime import datetime
            FW_NAME = Path(target[0].path).replace(Path(target[0].dir.path, datetime.now().strftime(new_name.replace('{date}', '%Y%m%d').replace('{time}', '%H%M%S.bin'))))
            print("FIRMWARE NAME: %s" % FW_NAME)
```

this is a quick fix / bandaid I came up with
add "*.bin*" to the end of `'{time}', '%H%M%S'`
have the `Path(...)` also equal FWNAME

thats really it. but im sure there is a formal way thats better.

thisiskeithb, 2023-06-10:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/25966

closed 2023-06-10 by thisiskeithb