MarlinFirmware/Marlin #25961: [BUG] Z-Fade Height causing sudden Z shift and failed prints

sargonphin, 2023-06-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The Z fade height feature causes a sudden change of Z height instead of progressively fading the bed leveling imperfections, causing a large gap in between the layers and failed prints. See the attached pictures below, the Z fade height was set to 12mm. This problem isn't new because I already had it with a previous version of the bugfix, although I don't remember which one exactly.

[image]
[image]
[image]
[image]

The obvious workaround is to set the Z fade height to zero so that it doesn't fade at all and doesn't cause these problems. We can get through with that because our bed doesn't have much height deviation (0.25mm round about) but for bend-o-matic thin aluminum sheet beds that can sometimes have over 1mm of difference between the edges, this could be a very big problem

### Bug Timeline

Sometime in the bugfix-2.1.x timeline

### Expected behavior

The Z fade height compensates the Z height difference in tiny steps instead of large steps at a time

### Actual behavior

The Z fade compensation was applied all at once in a single Z move

### Steps to Reproduce

1. Set the Z fade height to some height
2. Print
3. Observe sudden shifts in Z height in print at Z height corresponding to the Z fade height
4. Throw away the print & get mad :P 

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Modified Tronxy CoreXY

### Electronics

SKR 1.4 Turbo with TMC2209

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11712010/Configuration.zip)

A potential Feature Request is to be able to chose whether we want negative correction ("crushes" the high points) or positive correction ("stretches" the low points). Another extension to this feature request is that, Z fade height dynamically adjusts the extruded length according to the new layer height generated by the correction. Should I open new [FR] for that?

sargonphin, 2023-06-10:
This is what it looks like with Z fade height disabled (set to zero)

[image]

RV-from-be, 2023-06-17:
@sargonphin Sorry, but there is no issue with the Z FADE HEIGHT. In your case, when it is activated, the Auto Bed Leveling correction is applied to the first 12 mm. Then he is no longer. Your observed defects just indicate that you have either a problem with the geometric adjustment of your printer (bed, parallelism of the X axis, etc...), or the mesh (bed leveling) is not correct. Obviously, if you set your Z FADE HEIGHT to 0, it hides the geometry problem.

sargonphin, 2023-06-17:
The bed leveling works correctly and this is not how Z fade height works either ^_^ 

It is supposed to apply progressively less correction until the Z fade height is reached, flattening the X and Y axis. Setting it to 0 permanently applies the mesh leveling correction to the entire print.

The problem here is that it applies the Z correction incorrectly until the Z fade height is reached, from where it will apply nothing anymore and flatten the axis. This causes this shift in Z. It is sudden and large, looks like a couple layers. Sometimes, it prints in the air, several millimeters above the model.

DerAndere1, 2023-06-22:
Was this consistently reproducible when using the same gcode file? reproducible with a different gcode file?

sargonphin, 2023-06-23:
Yes this is reproductible using any gcode file, even twice the same in a row. I have checked the gcode and it's clean, there's no problem in it. It's firmware related, I got the problem twice after I posted the ticket because I forgot to write to eeprom

closed 2023-09-03 by github-actions[bot]

moswald14, 2023-09-13:
having this issue too. also my printer skips steps when canceling the print, which could be related. it messes up my sensorless homing somehow. I have to do a M502 after it skips steps or the homing sensitivity is too high. also a modified Tonxy x5s
[image]

moswald14, 2023-09-13:
update: enabling #define RX_BUFFER_SIZE 1024 has seemed to fix it? hopefully. I had this enabled before but thought it was the culprit lagging my screen but im currently 8.39% through a 19 hour print with no skipped steps