MarlinFirmware/Marlin #25963: [BUG] Marlin full size logo causes stuck on the updating screen

ilko-k, 2023-06-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Commenting out the **#define BOOT_MARLIN_LOGO_SMALL** line in order to use the full size logo causes the update process to stuck at 100%. Then, when the printer is turned OFF and ON, the screen remains black forever and not being detected via USB.

### Bug Timeline

New, noticed when customizing the fw

### Expected behavior

I expect the full sized logo being displayed at boot.

### Actual behavior

Stuck on the updating screen when it reaches 100%, then the printer doesn't boot when turned OFF/ON later on.

### Steps to Reproduce

1. Make sure you have defined **TFT_COLOR_UI** (and maybe _TOUCH_SCREEN_ too) in _Configuration.h_
2. Disable **BOOT_MARLIN_LOGO_SMALL** in _Configuration_adv.h_
3. Compile and try updating your printer

### Version of Marlin Firmware

bugfix_2.1*

### Printer model

Flying Bear Reborn 2

### Electronics

MSK Robin Nano v3 board, MKS TS35 v2 touchscreen, TMC2209 drivers

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[FBR2_Marlin_bugfix_2.1.X.zip](https://github.com/MarlinFirmware/Marlin/files/11712582/FBR2_Marlin_bugfix_2.1.X.zip)

ellensp, 2023-06-10:
Check the chip on the controller, make sure its a STM32F407VGT6 perhaps its a clone or a variant we don't know about with a chip with less resources.

ilko-k, 2023-06-10:
> 

Well, you were right, sadly.
[image].

I'm closing this. Sorry for the inconvenience and thanks for the help.

closed 2023-06-10 by ilko-k

ellensp, 2023-06-23:
@ilko-k you might want to try my latest PR.
It compressed the boot screen data, so full sized marlin boot screen will now fit on smaller  MPU

ilko-k, 2023-06-23:
@ellensp It looks promising, I'll try ASAP.

ilko-k, 2023-06-25:
@ellensp hey, excuse me but this feature doesn't seem to be available in the latest bugfix actually. What am I missing ?

ellensp, 2023-06-25:
At the moment its what is known as a pull request, the code is being being examined, cleaned up and prepared for inclusion to marlin bugfix.

T try it right now you would have to download my fork of marlin. Or wait a little it its added not bugfix

ilko-k, 2023-06-25:
oh so it's not merged yet, I see. Thanks.