MarlinFirmware/Marlin #25964: [BUG] Bed PID calibration heats extruder instead

apalazzi, 2023-06-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I'm trying to do PID calibration for the bed, but when I send the command:

M303 EBED S60 U1

the hotend gets warm instead and after some time the printer halts with an error.

### Expected behavior

The bed should heat and the PID calibration should be carried on.

### Actual behavior

The hotend started to warm up, then the printer stopped with an error.

### Steps to Reproduce

Send the command M303 E-1 S60 U1 to the printer.

### Version of Marlin Firmware

Bugfix-2.1.x

### Printer model

Tevo Tarantula

### Electronics

Stock

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Slic3r

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ilko-k, 2023-06-10:
You're doing it wrong, it should be "M303 E_index_ S60 U1", where index is the number corresponding to the extruder, (E0, E1...) and **E-1** for the heated bed. It doesn't use names.

closed 2023-06-10 by ellensp

ellensp, 2023-06-10:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

apalazzi, 2023-06-11:
If EBED has no meaning shouldn't the correct behaviour be to raise an error and not heat something else? At least this is what I would expect it to do.

EvilGremlin, 2023-06-11:
Such checks are meaningless, they'd only bloat firmware size for no reason.

apalazzi, 2023-06-11:
Sorry, but I disagree: if you're sending meaningless - and potentially harmful - command to your printer, this should be intercepted. This is not bloating the code, it's just good coding practice.
As for the wasted CPU cycles, my impression is that it's negligible, but I might be wrong. If you have data that shows otherwise, feel free to present it to me and I'll change my mind.
If you can point me to the relevant code I might also do some testing myself.

EvilGremlin, 2023-06-11:
This is intercepted by ignoring invalid data. Verbose erroring on ignorable stuff is extremely bad coding practice when you have only 128k ROM. And there are ways to actually do pysical damage with perfectly valid commands, and filtering these is downright impossible. All CNC machines for decades operated on assumption that operators know what they do. You could do filtering etc, but not on MCU. Such things are only appropriate in Klipper and/or Octoprint, and even then with restrictions.

apalazzi, 2023-06-14:
You seem to have missed my point.
I'm not asking verbose error reporting for everything, I'm just saying that if a bad command is sent it should not do something different and unexpected, hopefully it should be rejected with an error message, and ideally with a hintful error message. Sure, there are always ways to do physical damage to the machine with valid command, but if we can prevent it easily... why not?
I think that discarding an invalid command would not make the code bloated, and "we always did it like this" is not a good reason to not make things better; and the assumption that the operator knows what he's doing is plain wrong - everybody makes errors, you can´t rely on the operator always acting perfectly, it's simply not human.
I also think that returning a string like "invalid command" would be easy and cheap, but I'm not willing to spend hours in source diving just to prove my point: and again, if you can point me to the relevant source I might try to make a patch, and see for myself if it will really blot the code to an unbearable level.

EvilGremlin, 2023-06-14:
No i didin't. General "wrong command" fallback error indeed isn't expensive, and marlin already have error message for unknown gcode. But checking and erroring out for _parameters and values_ of each gcode command is very expensive. Relevant source - whole gcode directory. Benchmark system - mega1284 with 12864 LCD and SD.