MarlinFirmware/Marlin #25968: [BUG] ¿Configuration? SD Conflicts with SD Card on RAMPS 1.4

Amachete, 2023-06-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

SD Conflicts with SD Card on RAMPS 1.4.

I had TMC2130 and SDCARD,  working together, without problem. 

I Have just updated the firmware and the SDCARD stopped working and TMC2130 aswell.

When unplug the cable for SPI for the TMC drivers, marlin recognices the SDCARD and is able to read it.

 

### Bug Timeline

I do not know, It´s been a while without updatinf

### Expected behavior

TMC2130 and SD card worktogether without issues.

### Actual behavior

3D PRINTER is unable to work.

### Version of Marlin Firmware

latest

### Printer model

P3STEEL

### Electronics

RAMPS-MEGA

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin2.0_conf_2023.zip](https://github.com/MarlinFirmware/Marlin/files/11713430/Marlin2.0_conf_2023.zip)

[M43.txt](https://github.com/MarlinFirmware/Marlin/files/11713432/M43.txt)

It looks similar to:

[BUG] MKS Robin E3D v1.1 + TMC2130 SPI conflicts with SD Card SPI #22755

ellensp, 2023-06-11:
You don't actually give any details on how its wired

Current config is for a hardware spi on sdcard, but software spi on tmc2130's

but for eg the BTT ramps 1.6+ has hardware spi to all devices

what are you using?

ellensp, 2023-06-11:
If you are young software spi for your tmc's you need to enable #define TMC_USE_SW_SPI

Amachete, 2023-06-11:
Hello.

I use the same SPI pins (port) as the SD, I will include a Picture.

I have just seen the change that has messed up my set up..

The old RAMPS configuration you use the default SPI configuration, or use a SW-SPI PINS

AS from BUGFIX-2.0X:

pins_RAMPS.h

//
// TMC software SPI
//
#if ENABLED(TMC_USE_SW_SPI)
  #ifndef TMC_SW_MOSI
    #define TMC_SW_MOSI                       66
  #endif
  #ifndef TMC_SW_MISO
    #define TMC_SW_MISO                       44
  #endif
  #ifndef TMC_SW_SCK
    #define TMC_SW_SCK                        64
  #endif
#endif

On the new pins_RAMPS.h

The default  pins are:

//
// TMC software SPI
//
#ifndef TMC_SPI_MOSI
  #define TMC_SPI_MOSI                        66
#endif
#ifndef TMC_SPI_MISO
  #define TMC_SPI_MISO                        44
#endif
#ifndef TMC_SPI_SCK
  #define TMC_SPI_SCK                         64
#endif

So you have to write your SPI-pins...on the configuration.h

TO use the default SPI as the SD.

/**
   * Software option for SPI driven drivers (TMC2130, TMC2160, TMC2660, TMC5130 and TMC5160).
   * The default SW SPI pins are defined the respective pins files,
   * but you can override or define them here.
   */
  //#define TMC_USE_SW_SPI
  #define TMC_SPI_MOSI  51
  #define TMC_SPI_MISO  50
  #define TMC_SPI_SCK   52

So It ´s fixed for me, 

Thank you for your time.

closed 2023-06-11 by Amachete