MarlinFirmware/Marlin #25980: Random movement xy axis mks robin nano[BUG] (bug summary)

franmuzi1, 2023-06-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I compiled marlin for my neptune 2 (mks robin nano v 1,3), everything works except random axis movement during print.
it continue to extrude while randomly tracking lines on the buildplate.

### Version of Marlin Firmware

2.1.2.1

### Printer model

Neptune 2

### Electronics

E3d Hotend

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

](url)

franmuzi1, 2023-06-14:
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11747318/configuration.zip)

leftbrasil, 2023-06-23:
I'm experiencing this issue as well. I have a feeling that it have something to do with SD Card logic. Have you tried printing when there's single file only on the card?

franmuzi1, 2023-06-23:
Yes I tried, nothing changed.
Also tried replacing the SD not solving my problem

closed 2023-06-23 by franmuzi1

toomuchwonder, 2023-06-24:
I also have the same issue. I couldn't print anything successfully from the SD card. I've since moved to printing via serial port using Octoprint and still get occasional layer shifts in both X and Y directions.

RonnyLV, 2023-07-11:
+1 on the issue. Sometimes toolhead just wanders off in the middle of a print

RonnyLV, 2023-07-11:
@franmuzi1 you closed the issue - did you find a solution to the problem?

phobos79, 2023-07-27:
is a duplicate of this:
https://github.com/MarlinFirmware/Marlin/issues/25653

if you want to fix, install the 2.1.1.1, if you want input shaping on it i can give you the folder

thinkyhead, 2023-10-07:
Duplicate of #25653