MarlinFirmware/Marlin #25981: [BUG] SD Card not working

reynum2, 2023-06-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have installed a RAMPS 1.4 on my Anet A8 and installed Marlin 2.1.2.1 on it.
But when I insert my SD card I have the message "Media init fail"

I have just enable the 
"#define SDSUPPORT" 
and the 
"#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER"
The SD card is Fat32 and works well on another 3D printer
parameters
[image]

### Version of Marlin Firmware

2.1.2.1

### Printer model

Anet A8

### Electronics

RAMPS 1.4

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11748082/Configuration.zip)

ellensp, 2023-06-14:
Not all sdcard are equal

Not all 3d printer access sdcards in the same way

Many new modern sd card no longer support the legacy SPI protocol that a old ramps uses

New 3d printers with 32bit controllers tend to use SDIO protocol instead..

It is very likely your sd card just does not work using SPI

reynum2, 2023-06-27:
Hello !
Any update regarding this problem ?

reynum2, 2023-06-30:
Hello,

I tried to enable the parameter 
#define SDCARD_CONNECTION LCD

But it still not working

reynum2, 2023-06-30:
> It is very likely your sd card just does not work using SPI

What your are saying is Marlin uses SPI to communicate with the SD Card ?

I ask that because I just test the connection in accordance with :
[image]

And the connection with D49 -> D49 ( SD-Card detect ) works well 
But the connection with D49 -> D49 ( SD-Card SPI/MOSI ) does not work 
It way be a reason why my SD-Card reader does not work.

What do you think about ?

ellensp, 2023-06-30:
That picture is in error

see pins diagram 

https://reprap.org/mediawiki/images/c/ca/Arduinomega1-4connectors.png

the connector is

5v D49 
D50 D51  this is the MOSI pin
D52 D53
gnd NC

ellensp, 2023-06-30:
There is nothing wrong with marlin or your config

You have a hardware issue

I just tested this on my stock RAMPS 1..4 with REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

reynum2, 2023-07-02:
> I just tested your configs on Marlin 2.1.2.1 on my stock RAMPS 1..4 with REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
> SD card detects and reads fine

Thank you for this feedback.
Now I now where I should search to make this SD-Card works

I have a suspicion on the chip 74HC4050D just near the SD reader

[image]

ellensp, 2023-07-02:
That chip is being used as a voltage level snifter 5v to 3.3v   This is certainly a possibility. 

You can test this.. you disconnect the lcd   you use wires to connect 5v and gnd/  you connect 5v to the D51 sdcard mosi and check the output of the chip is 3.3v on the mosi pin , repeate for D52 sdcard sck and D53 sdcard cs    

But Ive seem many more issues with the ramps to lcd adapter board, with bad soldering

reynum2, 2023-07-02:
> But Ive seem many more issues with the ramps to lcd adapter board, with bad soldering

Year you're very good !
I found the mistake
[image]
Look at this sneaky little wrong solder
It just fixed it and everything works well now \o/