MarlinFirmware/Marlin #25985: [BUG] M20 Causes GT2560 V4.0 to crash

SparkyDan555, 2023-06-15:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Issuing M20 from any host (tested with ESP3D, Repetier-Host Mac, Pronterface) causes Marlin to crash and reboot. 

This bug does not occur when there is no SD card in the slot. Files can be read and executed from the LCD without issue. I have tested different SD cards with files, without files and freshly formatted which all cause the crash.

### Bug Timeline

Roughly last 2-3 weeks of bugfix (I think)

### Expected behavior

The SD card files are listed to the host. 

### Actual behavior

Marlin crashes and reboots.

### Steps to Reproduce

1. Insert SD card in slot
2. Connect host over serial (Baud 250000)
3. Issue M20 command (refresh files in ESP3D)
4. Observe boot screen on LCD and boot text in console.

### Version of Marlin Firmware

Marlin bugfix-2.1.x Jun 15 2023 12:10:37

### Printer model

Geeetech A10M

### Electronics

Stock GT2560 V4.0

### Add-ons

None

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Other (explain below)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Logs:
> 12:35:43 pm: N184 M20 *60
< 12:35:43 pm: Begin file list
< 12:35:43 pm: start
> 12:35:43 pm: N1 M110 *2
< 12:35:43 pm: Marlin bugfix-2.1.x  <------- Marlin crashes and reboots.

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/11757243/Archive.zip)

ellensp, 2023-06-15:
this is often caused by having lots of gcode files on your sdcard.  (mega2560 only has 8k of memory 
or a corrupted sdcard 
in both case reformat the sdcard to remove files and fix any corruption

SparkyDan555, 2023-06-15:
> 

This is why I formatted the card and tried multiple other cards that are working on other machines. Unfortunately this did not fix the issue.

SparkyDan555, 2023-08-15:
This is still an issue in bugfix-2.1.x.

fausby, 2023-10-23:
Me too on GEEETECH A20T GT2560 4.1 B

fausby, 2023-10-23:
Testing, I found that if I disable BLTOUCH, as provided in the example files, with PROBE_MANUALLY, then it works.
¿WHY?

SparkyDan555, 2023-10-25:
> Testing, I found that if I disable BLTOUCH, as provided in the example files, with PROBE_MANUALLY, then it works. ¿WHY?

I have tested disabling BLTouch (and temporarily enabling FIXED_MOUNTED_PROBE to satisfy sanity checks) and it still crashes when M20 is sent from a host.

fausby, 2023-10-29:
I have GT2560 4.1B, with default configuration files the SD works.  With BLTOUCH, when I insert the SD, it reboots until I remove the SD.

fausby, 2023-10-30:
OK, if my SD has few files, it works.

closed 2024-06-15 by github-actions[bot]