MarlinFirmware/Marlin #25990: White screen after boot elegoo neptune 2

franmuzi1, 2023-06-17:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

tried to comment // #define TFT_RESET_PIN                     PC6   // FSMC_RST
on mks robin nano common.h but nothing changed

### Version of Marlin Firmware

2.1.2.1 bugfx

### Printer model

neptune 2

### Electronics

stock

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/11778573/config.zip)

jmz52, 2023-06-18:
```
  #define TFT_DRIVER ILI9341
  #define TFT_INTERFACE_FSMC
  #define TFT_RES_480x320
```
Invalid configuration, you are using wrong model for TFT controller.
Max screen resolution for ILI9341 driver controller is 320x240.

EvilGremlin, 2023-06-18:
This slipped into example conf... and used to work fine at some point i suppose. Gotta be MKS_ROBIN_TFT35 i guess.