MarlinFirmware/Marlin #25993: [BUG] (E max temp not respected on restart)

landodragon141, 2023-06-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

My max E temp is set to 300C. When E-stopping with the extruder at 255 (printing ABS) marlin throws the error "E1 Max Temp". I only have 1 extruder so shouldn't this be E0?

### Steps to Reproduce

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/11780315/Marlin.zip)

### Version of Marlin Firmware

Latest Bugfix and 2.1.2.1

### Printer model

Custom

### Electronics

SKR 1.3

### Add-ons

PT1000?

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-06-18:
Displays start at E1, board and code internally use E0. 
If you really want it to say E0 there is a setting for it

ellensp, 2023-06-18:
'When E-stopping"  how are you imitating this?

Please fill in the steps to reproduce.

landodragon141, 2023-06-19:
I'm using an emergency stop button on my display. The same problem occurs if I just shut of the power and turn it back on. It seems like it needs to get down to 245 ish before it will fully start

thinkyhead, 2023-11-05:
Seems interesting. If you're able to try the latest `bugfix-2.1.x` with all your other optional features disabled, that could help to isolate whether this is something with the UI or the startup or the temperature. At the moment it sounds like there hasn't been any major attempt to isolate the cause, so we should get on that.

landodragon141, 2023-12-22:
Sorry I've been crazy busy and still haven't had a chance to get to testing this.....

closed 2024-09-23 by github-actions[bot]