MarlinFirmware/Marlin #25994: [Problem] Marlin-2.0.9 just turned on the FLASH_EEPROM_EMULATION ，The serial port of the motherboard no longer prints

Wenuo, 2023-06-18:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

I used STM32F103RCT6 as my own printer motherboard and turned on the FLASH_EEPROM_EMULATION function, but the serial printing of the motherboard is stuck and the screen keeps restarting；
I found that this function is related to the firmware size, when I comment out the #define OLED_PANEL_TINYBOY2, the firmware size is only 162164 bytes, at this time the motherboard printing is normal, and the analog eeprom is also normal；
but after I turn on the #define OLED_PANEL_TINYBOY2 function, the firmware size becomes 223356 bytes, and the serial port printing is not normal at this time;

Version 2.0.9.1 behaves the same as version 2.0.4.4；
My veneer doesn't have a bootloader，just use st-link upload；I calculated that the firmware does not conflict with the flash size；
Hope someone replies！

### Expected behavior

As depicted in the picture, the simulated eeprom loads normally；
[image]

### Actual behavior

In fact, when the firmware is uploaded to the MCU through stlink, the serial port printing is stuck, and the simulation eeprom seems to fail；
[image]

### Steps to Reproduce

Turn on macro definitions, make the firmware larger, and you can reproduce it；

### Version of Marlin Firmware

2.0.9.1/2.0.44

### Printer model

ender-3

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11780601/Configuration.zip)
[pins_.zip](https://github.com/MarlinFirmware/Marlin/files/11780602/pins_.zip)

Wenuo, 2023-06-18:
**Supplement the compilation environment**
[platformio]
src_dir = Marlin
boards_dir = buildroot/share/PlatformIO/boards
default_envs = STM32F103RC
include_dir = Marlin

[env:STM32F103RC]
platform = ${common_stm32f1.platform}
extends = common_stm32f1
board = genericSTM32F103RC
platform_packages = tool-stm32duino
monitor_speed = 115200
board_build.offset = 0x0000
board_upload.offset_address = 0x08000000
lib_deps = marlinfirmware/U8glib-HAL@^0.5.2

thisiskeithb, 2023-06-18:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

Wenuo, 2023-06-18:
> Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

The phenomenon is the same,Positively correlated with firmware size.
[image]
The serial port will no longer be printed at the red line mark！

thisiskeithb, 2023-06-18:
>  The serial port will no longer be printed at the red line mark！

Did you reset your EEPROM with `M502` followed by `M500`?

Wenuo, 2023-06-18:
> > The serial port will no longer be printed at the red line mark！
> 
> Did you reset your EEPROM with `M502` followed by `M500`?

Since the serial port is stuck, I can't enter M502, M500

ellensp, 2023-06-18:
enable   #define EEPROM_AUTO_INIT  // Init EEPROM automatically on any errors.

thisiskeithb, 2023-06-18:
You can also initialize EEPROM through the LCD by going to:

Configuration -> Advanced Settings -> initialize EEPROM

You can also enable `EEPROM_AUTO_INIT` & `EEPROM_INIT_NOW`:

https://github.com/MarlinFirmware/Marlin/blob/dd1792bdfce1ff5ff679f20dbf9bf66789e743b0/Marlin/Configuration.h#L2303-L2304

Wenuo, 2023-06-18:
> You can also initialize EEPROM through the LCD by going to:
>
> Configuration -> Advanced Settings -> initialize EEPROM
>
> You can also enable `EEPROM_AUTO_INIT` & `EEPROM_INIT_NOW`: [...]
I don‘t think ’EEPROM  VERSION MISMATCH‘ is a porblem；I mean that as the firmware gets bigger and bigger, the serial port is easy to get stuck and even restarts all the time；

Wenuo, 2023-06-18:
Thanks to everyone for their support, I added an external EEPROM with IIC  and it works fine.

closed 2023-06-18 by Wenuo