MarlinFirmware/Marlin #25998: [BUG] Rapid Undefined Motion On Print Start

Gryd3, 2023-06-19:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

Printer makes a very rapid movement (faster than set with M203) when starting a print.
Direction of the rapid movement (for me) is in the positive XY, negative Z direction, and unknown E direction. (Away from the end-stops and into the print bed.)
This issue seems to be caused by running a specifically crafted gcode file that ends with an M501, and was encountered in the wild running a default generated code from elis3dp.com for Linear Advance tuning. After running the 'problem file' the Z height generally reports a negative Z value, or a very large (800) positive Z value.  (Please note, soft end stops are enabled, and machine size is roughly 220mm cubed.) Please also note that the 'problem file' executes fine with no issues during run-time.  It's only after (or at the end) of the file that a problem arises, as it seems to put the machine into an unstable state once complete.
The problem is 'triggered' after being set by the 'problem file' by what I assume is G28, but I have not done extensive testing to see if it's any motion command, non-motion command, G28 itself, or simply the act of starting a new file.  (Observed with SD and printing from USB with OctoPrint) The 'trigger' can be any other print file, as long as a problem file was ran immediately before-hand.

Recreating the problem:
1) Run the pa_pattern or a 'BorkedCode' file.  After running, please note the reported Z height in the UI. (If displayed value is incorrect, the machine may be in 'glitched state'.  *Machine may be glitched with no indicator!
2) Run any other file, machine's first motion will be rapid on XYZ and E axis. Be ready to disable the printer before it crashes.

My Environment: 
- Ender 3 Neo, firmware attached. Marlin 2.1.2 from late April.  Have not tested latest 2.1.2 yet.
*Firmware and configs attached in ZIP. 
- Ender 3 running on Creality V4.2.2 with GD32F303RET6 and TMC2208 (E Stepper OTP'd to disable SilentStep... was required for me to run LinAdv on Marlin 2.0.9.5)
- May be run from USB or SD.  Different SD cards tested. 

Pending Tests:
- Does the glitch occur on my Ender 3 Pro?
- Does the glitch occur on a newer 2.1.2 build, or the 2.0.9.x build?
- Shrinking the 'BorkedCodev3' file further to identify which GCode lines cause the issue ** Note that pasting the last 6 lines from this file to another file has not appeared to cause the glitch

Disclaimer and reason for report. 
The undefined and sudden motion may be a danger on larger machines running Marlin. I do not have the capacity to determine the 'Exact Cause' of this problem, but as of now can work-around this problem by removing the M501 from the end of the file. The M501 is not a normal work-flow item for me, but 'could be' for others who make in-memory changes to the machine in print files and desire a known state once the file finishes. While I would be happy leaving this as-is, if this problem persists in other firmware, awareness and a fix are strongly suggested.   Please mark as closed if this issue cannot be recreated. 

[BorkedCodev3.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/11790776/BorkedCodev3.gcode.txt)
[pa_pattern_0-50.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/11790777/pa_pattern_0-50.gcode.txt)
[BorkedCodev1.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/11790778/BorkedCodev1.gcode.txt)
[2.1.2-G1.0.3 (E3Neo).zip](https://github.com/MarlinFirmware/Marlin/files/11790792/2.1.2-G1.0.3.E3Neo.zip)

### Bug Timeline

2023-04

### Expected behavior

I expected the machine to operate as expected, which in the case of my tests, would have been a proper execution of G28 (home) by moving to the X/Y end-stops, before centering and moving to the Z end-stop.

### Actual behavior

The machine moved very rapidly on the opposite direction of the X/Y end-stop colliding with mounting brackets at the end of the physical limit, ground the filament by prematurely moving the E axis, and breaking the CR-Touch by slamming the Z axis down into the table as it made a lateral move.

### Steps to Reproduce

Recreating the problem:
1) Run the pa_pattern or a 'BorkedCode' file.  After running, please note the reported Z height in the UI. (If displayed value is incorrect, the machine may be in 'glitched state'.  *Machine may be glitched with no indicator!
2) Run any other file, machine's first motion will be rapid on XYZ and E axis. Be ready to disable the printer before it crashes.

### Version of Marlin Firmware

2.1.2

### Printer model

Ender 3 Neo

### Electronics

Stock equipment, modified OTP bit on TMC2208 E Stepper Driver. (Board V4.2.2)

### Add-ons

CR-Touch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Slicer Detail:
 - 'Problem file' crafted with an online generator.  Not enough known about the problem file to craft an appropriate 'end-gcode' where this could be problematic for any other slicer.
 - 'Trigger file' has been made with the online generator, Cura, or SuperSlicer.  (The important aspect is the 'problem file' that puts the machine into an unstable state that is triggered by what is possibly 'any subsequent file)

Host Software:
 - Octoprint was first instance of problem. (Assumed bad host)
 - SD Card was first 'observed' instance of problem.
 **Problem was difficult to diagnose, as the damage caused to the CR Touch left me in a state unable to 'test'.. however the most recent issue was observed and reacted to quickly enough that I could remove power from the machine to carry out additional tests. Additionally.. the problem was not understood to be caused by a 'previously ran' file that appeared to function well.

Gryd3, 2023-08-19:
Hopeful someone else has attempted to test this. The motion type is concerning, and all files to attempt this have been attached above.

thisiskeithb, 2023-08-19:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

SerGicson, 2023-09-02:
> Загрузите [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip), чтобы протестировать последнюю версию кода, и сообщите нам, если проблема не устранена.
I have the same problem with Kingston KP3S in the latest version of the bugfix-2.1.x

SerGicson, 2023-09-02:
> Загрузите , чтобы протестировать всю версию кода, и сообщите нам, если проблема не устранена.[`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip)

I have the same problem with Kingston KP3S in the latest version of the bugfix-2.1.x

RJSDevel, 2023-09-03:
I've fixed this for Kingston KP3S Pro with code from https://github.com/MarlinFirmware/Marlin/issues/22898#issuecomment-1175972115 and uncomment #define DEBUG_LEVELING_FEATURE

SerGicson, 2023-09-04:
Спасибо, но функция:  bool SDIO_Init() сейчас выглядит не так как раньше, и я не понимаю что мне делать. 

  
>Воскресенье, 3 сентября 2023, 21:56 +03:00 от Yagupov Ruslan ***@***.***>:
> 
> 
>Я исправил это для Kingston KP3S Pro с помощью кода  #22898 (комментарий) и раскомментировал #define DEBUG_LEVELING_FEATURE.
>— 
>Ответьте на это письмо напрямую,  просмотрите его на GitHub или  отмените подписку . 
>Вы получили это, потому что оставили комментарий. Идентификатор сообщения:  <MarlinFirmware/Marlin/issues/25998/1704375551  @  github  .  ком> 
 
 
--
Сергей Славянин
Отправлено из Почты  Mail.ru

closed 2023-11-18 by github-actions[bot]