MarlinFirmware/Marlin #26000: [BUG] Failure to compile in VSCode after making any changes to Configuration.h and Configuration_adv.h Files

WyattNotEarp, 2023-06-20:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I'm trying to configure Marlin-Bugfix-2.1.X for a BTT SKR Mini E3 V3 Board. I've got a lot of mods done to this printer and it requires a lot of changes to Configuration.h & Configuration_adv.h. If I simply compile the firmware without making any changes, it results in success. However, once a single change is made, I get a long list of errors  pertaining to BOTH and EITHER statements. My research leads me to believe its a failure to define EITHER and BOTH in Marlin/src/core/macros.h. In the macros.h file placing #define EITHER(V1,V2)       ANY(V1,V2) and recompiling, results in only having errors for BOTH statements shown below. Can Someone show me a way to define BOTH(V1,V2) in that folder that could result in a successful compile. Perhaps this would be a worthwhile addition to the folder if this is the actual problem. I greatly appreciate your time.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11795162/Configuration.zip)

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/11795163/Configuration_adv.zip)

[macros.zip](https://github.com/MarlinFirmware/Marlin/files/11795165/macros.zip)

Some of the changes I made
1. Any typical for setting up a probe for the 5 pin connector and its respective pins. Probe_Z_Babystepping, end stop changes, stuff like that.
2. Adjusted the currents for the oversized motors I have on all of my axis( so I don't have to do this again later ). 
3. Adjusted E-steps.
4. Hot End Max temp 315C
5. Thermistor type.
6. Probably some other things as well

Errors while trying to configure,
[image]

Placing #define EITHER(V1,V2)       ANY(V1,V2) on line between #define ANY and #define All will remove all of my missing binary operator before token "(" problems on EITHER statement lines.
[image]

### Bug Timeline

First time compiling marlin. No prior experience.

### Expected behavior

Successful Compile

### Actual behavior

Failure to Compile

### Steps to Reproduce

1. Open VSCode
2. Auto build marlin with marlin-bugfix-2.1.x with Creality/Ender-3 Pro/BigTreeTech SKR Mini E3 3.0 files transfered over. 
3. Make changes
4. Build Marlin

### Version of Marlin Firmware

Marlin-Bugfix-2.1.X

### Printer model

Creality Ender 3 Pro

### Electronics

BTT SKR Mini E3 V3.0

### Add-ons

TFT35, E3D V6, CR Touch Probe, Belted Z,

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-06-20:
EITHER and BOTH macro was removed from Marlin bugfix after users complained there are to many macros that do that same thing.

Your Configs are out of date

replace BOTH with ALL
replace EITHER with ANY

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2023-06-20 by ellensp

WyattNotEarp, 2023-06-20:
I'll remember that in the future. I have no PC coding experience yet but as much similarity as it has with Fanuc Macro B, hopefully one day I'll actually get to help you guys out. Thanks for the quick and thorough reply.

nicwilson58, 2023-07-30:
This makes absolutely no sense whatsoever,   they are macros,  so what if they are they.  Ever heard of backward compatibility?  instead you waste time on such trivial things constant renaming defines,   removing macros.   It escapes me that you programmers have so much time on your hands to waste it on such trivial nonsense.

thinkyhead, 2023-10-06:
@nicwilson58 — You could simply ask why the change was made rather than attacking us. If you would like to understand the rationale for this change, there is a discussion on the Marlin Discord about it. If you disagree with this change, please add your feedback to that discussion.