MarlinFirmware/Marlin #26001: [BUG] (bug summary)

j0hn1k, 2023-06-20:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When printing, the printer spontaneously, in an arbitrary place, can start to park the extruder and then the board reboots. this is not always the case, about 1 case per 3-4 prints for 8-9 hours, there is no filament detector on the printer, OCTOPUS MAX EZ board, 5160 - X and Y drivers, the rest are 2209, parking without limit switches, BTT mini 128x64 screen, CORE XY, the printer is as simple as possible. there is also a UPS, I exclude power outages

### Bug Timeline

new

### Version of Marlin Firmware

02010300

### Printer model

BR

### Electronics

BTT OCTOPUS EZ MAX

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Desktop.zip](https://github.com/MarlinFirmware/Marlin/files/11801394/Desktop.zip)

j0hn1k, 2023-06-20:
I also have problems connecting the ESP32E module. (does not see the board) although the same module works fine with the SKR3 board

with the settings according to the BTT manual #define SERIAL_PORT_3 7 does not compile (I tried it on the raspberry version from the instructions from the BTT repository b c of this repository). And #define SERIAL_PORT_3 3 the module itself does not see the board

ellensp, 2023-06-20:
the esp32 pins documentation is contradictory

on https://github.com/bigtreetech/Octopus-Max-EZ it says

ESP-RX(UART7) | PE7 | 58
ESP-TX(UART7) | PE8 | 59

but on https://github.com/bigtreetech/Octopus-Max-EZ/blob/master/Hardware/BIGTREETECH%20Octopus%20MAX%20EZ%20V1.0%20PinOut.png
It says

PD8 USART3_TX
PD9 USART3_RX

And no Circuit diagrams, so cannot check those ....

ellensp, 2023-06-20:
Ive creating and issue on btt site https://github.com/bigtreetech/Octopus-Max-EZ/issues/16

j0hn1k, 2023-06-20:
ellensp, try to change the pins in the firmware of the board? the rest of this does not break there

j0hn1k, 2023-06-21:
What else can cause the extruder to park, besides pausing the filament change, or directly the g-code of the command?

closed 2023-09-05 by github-actions[bot]