MarlinFirmware/Marlin #26005: [FR] Almost all new Creality boards ship with GD32 processors. Please add official support.

Wreck-3D, 2023-06-22:
### Is your feature request related to a problem? Please describe.

The various versions of the Ender 3 are the most common printer on the market by far, and the new models come with a GD32F303RET6 SOC instead of the STM32. Even BTT is using G32 processors on their boards now. Despite this, Marlin still does not support these boards. Can this please be fixed? I admit I'm far from an expert on coding, but it seems it would be fairly easy to do (I may be mistaken).

### Describe the feature you want

Support for **GD32F303RET6** on Creality boards (such as the 4.2.2). Currently Marlin does not work on these boards. Some people reported getting it to work with STM32 Maple, but this does not seem to always work (did not work for my printer).

thisiskeithb, 2023-06-22:
Duplicate of https://github.com/MarlinFirmware/Marlin/discussions/24985

closed 2023-06-22 by thisiskeithb

Wreck-3D, 2023-06-22:
This is not a duplicate of #24985. For one, the boards were rare then, now GD32's are common, as in the #1 board shipping today. 2. he was requesting GD32F103RET6, this is a request for GDF**3**03RET6. Similar boards, but not the same. You are essentially just trying to silence requests for this. Why are you so against adding support for this? And you wonder why you struggle to even get 5 sponsors of Marlin?

thisiskeithb, 2023-06-23:
> You are essentially just trying to silence requests for this. 

We’re not silencing anyone. Please use that thread as the single place to discuss GD32 support. We don’t need multiple threads asking for every individual variant.

> Why are you so against adding support for this?

We’re not. Until someone submits a PR to add support, they won’t be fully supported.

wclearyf5, 2023-06-23:
A ? For programmers. Can and how do I change the firmware(g codet prevent the Zaxis home from lifting 10 mm after homing? Just asking. I want to learn tySent from my Verizon, Samsung Galaxy smartphone