MarlinFirmware/Marlin #26009: LA impresora se reinicia constantemente luego de actualizar a la ultima version de marlin [BUG] (bug summary)

andystoc, 2023-06-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

La impresora se reinicia constantemente luego de actualizar a la ultima version de marlin. Tiene puesta una placa skr 2 f429

### Bug Timeline

reinicio constante

### Expected behavior

.

### Actual behavior

.

### Steps to Reproduce

.

### Version of Marlin Firmware

2.1.2.1

### Printer model

artillery x1

### Electronics

skr 2

### Add-ons

bltouch 

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/11843562/Marlin.zip)

ellensp, 2023-06-23:
English version: Printer keeps rebooting after upgrading to the latest version of marlin

### Bug Description

Printer keeps rebooting after upgrading to the latest version of marlin. It has a skr 2 f429 plate on it

rq3, 2023-06-23:
Sounds just like my bug report #25852.

eoyilmaz, 2023-07-17:
Is that rebooting on startup or rebooting while printing?

rq3, 2023-08-03:
> Is that rebooting on startup or rebooting while printing?

Re-boots every 3 seconds on start-up.

thinkyhead, 2023-08-07:
Duplicate of #25852

closed 2023-08-07 by thinkyhead