MarlinFirmware/Marlin #26010: Incorrect temperature scaling of AD595 on 3.3 volt boards like BTT SKR 3 [BUG] (bug summary)

Mark4004, 2023-06-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The temperature reading from an AD595 thermocouple amplifier connected to a K type thermocouple (Temperature sensor type -1) reads high by (5 / 3.3 * actual temperature) on a controller board with a 3.3 volt ADC reference. For instance, 25C will display as 38C. This sensor works perfectly on a 5 volt board.

The problem is located in temperature.cpp line 1974 in 2.1.2.1 or line 2145 of bugfix-2.1.x
```
`#define TEMP_AD595(RAW)  ((RAW) * 5.0 * 100.0 / float(HAL_ADC_RANGE) / (OVERSAMPLENR) * (TEMP_SENSOR_AD595_GAIN) + TEMP_SENSOR_AD595_OFFSET) `
```
The "5.0" is the ADC reference voltage for a 5 volt board like the Arduino MEGA 2560. When replaced with a board like the BTT SKR 3 which uses a 3.3 volt ADC reference, the temperature reads incorrectly.

To fix the problem I changed the line to this:
```
`#define TEMP_AD595(RAW)  ((RAW) * 3.3 * 100.0 / float(HAL_ADC_RANGE) / (OVERSAMPLENR) * (TEMP_SENSOR_AD595_GAIN) + TEMP_SENSOR_AD595_OFFSET)`
```
The ideal fix isto have a parameter that defines the ADC reference voltage for each board instead of having to hard code it in temperature.cpp

### Bug Timeline

Discovered by replacing Arduino MEGA 2560 with BTT SKR 3

### Expected behavior

A temperature of 25.0C should display reasonably close to 25.0C 

### Actual behavior

25.0C displays as 37.6C.

### Steps to Reproduce

Connect a AD595 with thermocouple (Temperature sensor type -1) to input E0. 
For a temperature of 25.0C the display will show 37.6C. If you multiply 37.6 * (3.3/5) you get 24.8 which is close to 25.0 with only 10 bits of ADC resolution.

Instead of connecting an actual thermocoupe to E0, you can simulate it by applying 0.250 V to E0 thermister input.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11844388/Configuration.zip)

### Version of Marlin Firmware

2.1.2.1 but exists in all versions of Marlin

### Printer model

Custom Prusa i3 mk2 style

### Electronics

 BTT SKR 3 board

### Add-ons

AD595 thermocouple amplifier and E3D K type probe

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

A similar problem exists for anyone using an AD8495 on a 5 volt board because the 3.3 volt reference is hard coded on the line below the AD595.
```
`#define TEMP_AD8495(RAW) ((RAW) * 6.6 * 100.0 / float(HAL_ADC_RANGE) / (OVERSAMPLENR) * (TEMP_SENSOR_AD8495_GAIN) + TEMP_SENSOR_AD8495_OFFSET)`
```
The "6.6" is actually 2 * 3.3 volts. I was done that way because the AD8495 output is only 5mv/C compared to 10mv/C on the AD595. Like the issue I described above, the ADC reference voltage should be defined in the board data instead of being hard coded.
/

thisiskeithb, 2023-06-23:
This has come up a few times and in https://github.com/MarlinFirmware/Marlin/issues/11658, adding a config define like `ADC_REFERENCE_VOLTAGE` was suggested:

```cpp
#define ADC_REFERENCE_VOLTAGE <value>
```

GMagician, 2023-06-26:
I'm not sure how these device works/are connected but each HAL already have HAL_ADC_VREF

thinkyhead, 2023-06-28:
Thanks for reviving this question. It was not resolved clearly the last time it was raised and no one submitted the simple patch until now. If we can be assured that the ADC reference voltage is also the one used to supply voltage to the thermocouple amplifier then we can just use `HAL_ADC_VREF`. It is however theoretically possible to run a different voltage to the the amplifier, but that would be weird, since the result must be read by the ADC.

I'll do a quick review on the PR and make sure it's easier to see that for a 5V input the 8495 returns a scale of 5mV/°C and the 595 returns a scale of 10mV/°C. The current use of `(HAL_ADC_VREF) * 2 * 100` for the finer scale doesn't really make that very clear. The closer we can get to reflecting the maths described in the datasheets, the better.

thinkyhead, 2023-06-28:
According to the datasheets, you cannot even use an [AD595](https://www.analog.com/media/en/technical-documentation/data-sheets/AD594_595.pdf) with a 3.3V supply voltage; the minimum is 5V. For a 3.3V ADC you must use [AD8495](https://www.analog.com/media/en/technical-documentation/data-sheets/ad8494_8495_8496_8497.pdf) instead, since it allows for an input as low as 2.7V.

Do we know that to be the case for all AD595 amplifiers, or is that just a limitation of the AD595 that I linked to?

closed 2023-06-28 by thinkyhead