MarlinFirmware/Marlin #26022: [BUG] PWM not working on SKR Mini E3 V2

HeartOfGermany, 2023-06-27:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

With Marlin 2.1 (main branch) I cannot get any PWM working except on the parts cooling fan. I also tried DBP's marlin 2.0.9.1 Mesh Bed Leveling package, that behaved identically.

The SKR Mini E3 V2 uses the STM32F103 RCT6. It supports PWM on almost all pins on a hardware level according to the datasheet, as most pins have a channel of the hardware timers available. Specifically I want the pins PC12 and PC13 to output PWM. PC12 is configured as the SPINDLE_LASER_PWM_PIN. PC13 is defined as SPINDLE_LASER_ENA_PIN. The enable-pin is working correctly, but this is expected, as it is only on or off.

I have enabled to set pins, which are firmware protected and verified, that this even works with the Thermistor of the Bed (I disabled Bed protection). However I only get PWM out of PC6 (Parts Cooling Fan). All other pins set to 3.3 (or 5V for some pins) for any S value at or above 1.

I have tested the following pins to output PWM:
PA1 (Should work = Servo of Z-Probe)
PA2 (= TX2 - worked digitally at least, after disabling the serial port in the firmware, as it is unused)
PA3 (= RX2)
PA8 (Should work = LED Communication)
PC12 (= PS_ON - unused > reused for Laser PWM)
PC13 (= PT_DET - unused > reused for Laser Enable - works, as it is just on off)

Setting pins, even protected pins to on or off works with no exception, however PWM does not work. I did not adjust PWM settings, as I thought on the 2.0.9.1 DBP fork the issue was, that I changed the PWM frequency and used FAST_PWM_FAN. This does not appear to be the case. What might be the issue here?

### Bug Timeline

2.0.9.1 > Not tested before

### Expected behavior

"M42 I P38 S127 T1" causes PC6 Parts cooling Fan to output 50/50 PWM
"M42 I P44 S0 T1" causes PC12 to output 0/100 PWM (Off)
"M42 I P44 S127 T1" causes PC12 to output 50/50 PWM
"M42 I P44 S255 T1" causes PC12 to output 100/0 PWM (On)

### Actual behavior

"M42 I P44 S127 T1" causes PC12 to output 1 (3.3V)

"M42 I P38 S127 T1" causes PC6 Parts cooling Fan to output 50/50 PWM (~1.15V on controller pin directly)
"M42 I P44 S0 T1" causes PC12 to output 0 (0V)
"M42 I P44 S127 T1" causes PC12 to output 1 (3.3V) <--- This is the issue
"M42 I P44 S255 T1" causes PC12 to output 1 (3.3V)

### Steps to Reproduce

Change Board to the skr mini e3 v2
Enable the Laser Feature and set PWM pin to PC12 and Laser Enable to PC13
Disable firmware protection and enable manual pin control via M42 (As for other projects I absolutely need some more pins and there is no other choice)
Compile
Try to create a test laser pulse

### Version of Marlin Firmware

Marlin 2.1

### Printer model

Highly Frankenstein-ized Ender 3

### Electronics

SKR Mini E3 V2

### Add-ons

5W Laser with Relay PCB for laser Enable and PWM signal amplifier for 0/12V PWM instead of 0/3.3V PWM from the microcontroller.

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

None

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

No other changes except what is described above.

ellensp, 2023-06-27:
these are the only PWM pins 

  {PA_0,  TIM2,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_NONE, 1, 0)}, // TIM2_CH1
  {PA_1,  TIM2,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_NONE, 2, 0)}, // TIM2_CH2
  {PA_2,  TIM2,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_NONE, 3, 0)}, // TIM2_CH3
  {PA_3,  TIM5,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_NONE, 4, 0)}, // TIM5_CH4
  {PA_6,  TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_NONE, 1, 0)}, // TIM3_CH1
  {PA_7,  TIM8,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_NONE, 1, 1)}, // TIM8_CH1N
  {PA_8,  TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_NONE, 1, 0)}, // TIM1_CH1
  {PA_9,  TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_NONE, 2, 0)}, // TIM1_CH2
  {PA_10, TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_NONE, 3, 0)}, // TIM1_CH3
  {PA_11, TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_NONE, 4, 0)}, // TIM1_CH4
  {PA_15, TIM2,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_TIM2_PARTIAL_1, 1, 0)}, // TIM2_CH1
  {PB_0,  TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_NONE, 3, 0)}, // TIM3_CH3
  {PB_1,  TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_TIM1_PARTIAL, 3, 1)}, // TIM1_CH3N
  {PB_3,  TIM2,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_TIM2_PARTIAL_1, 2, 0)}, // TIM2_CH2
  {PB_4,  TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_TIM3_PARTIAL, 1, 0)}, // TIM3_CH1
  {PB_5,  TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_TIM3_PARTIAL, 2, 0)}, // TIM3_CH2
  {PB_6,  TIM4,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_NONE, 1, 0)}, // TIM4_CH1
  {PB_7,  TIM4,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_NONE, 2, 0)}, // TIM4_CH2
  {PB_8,  TIM4,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_NONE, 3, 0)}, // TIM4_CH3
  {PB_9,  TIM4,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_NONE, 4, 0)}, // TIM4_CH4
  {PB_10, TIM2,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_TIM2_PARTIAL_2, 3, 0)}, // TIM2_CH3
  {PB_11, TIM2,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_TIM2_PARTIAL_2, 4, 0)}, // TIM2_CH4
  {PB_13, TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_NONE, 1, 1)}, // TIM1_CH1N
  {PB_14, TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_NONE, 2, 1)}, // TIM1_CH2N
  {PB_15, TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_NONE, 3, 1)}, // TIM1_CH3N
  {PC_6,  TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_TIM3_ENABLE, 1, 0)}, // TIM3_CH1
  {PC_7,  TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_TIM3_ENABLE, 2, 0)}, // TIM3_CH2
  {PC_8,  TIM8,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_NONE, 3, 0)}, // TIM8_CH3
  {PC_9,  TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_TIM3_ENABLE, 4, 0)}, // TIM3_CH4

Pc12 and pc13 is not in this list

HeartOfGermany, 2023-06-27:
> these are the only hardware capable PWM pins
> ...
>  {PA_8, TIM1, STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, AFIO_NONE, 1, 0)}, /
> ...
> PC12 and PC13 is not in this list

But why is PA 8, which is Servo 0 not outputting PWM but only 0 and 1? It is actually required to output some form of analog signal to control a servo correctly.

Thank you so far. I was absolutely not able to find these lines in the 2.0.9.1 version by DBP - I am confused as to why the Cooling fan was even able to be set in an analog way without the definition of the pinmap `WEAK const PinMap PinMap_PWM[] = {...`

ellensp, 2023-06-27:
provided config is 02010201 

Please test the [`bugfix-2.1.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to see where it stands.

HeartOfGermany, 2023-06-27:
@ellensp Yes, as 2.1.2.1 is what I use, as it is in release state which I typically try to use. I do not know, what mistake I made before, but it now works for PWM pins with M42 as edited above. Thank you. The provided config indeed allows FAN0 and FAN1 to be controlled linearly, so it might be a good reference for people having this issue later.

closed 2023-09-07 by github-actions[bot]

HeartOfGermany, 2023-09-19:
Just a follow up. I am not really considering this issue fixed actually. I am able to control the pin state(s) with M42 command with Analog output.

- But it is not PWM
- I am unable to use it with M4 I S# command at all. No matter, what I try, i cannot seem to make it work. Is anyone able to say, what this depends on or how I could debug this issue?

I am happy with the analog output, as I have a PWM generator hooked up fixing this issue. The issue is, that M42 is just not working reliable enough. The timing is off all the time, no matter what type of gcode magic I use. For example splitting up the gcode in 0.05mm sections actually works to get the timing of each pulse related to each other better, but for bidirectional engraving this is useless, as from the midpoint it is shifted about a centimeter to the travel direction because of the delay. Engraving unidirectionally shifts it just in one direction, but the timing is not the same for each line, making a line 90° to the printing direction just a dotted mess. Unusable.
Both the laser PWM pin and the pin I set with M42 somewhat successfully is PA1.

So is there a way to make this possible at all? I cannot find anyone doing this engraving with the SKR Mini E3 V2 at all. Others just write about basic line engraving for cutting. But I need a way to engrave images.

Tl;Dr:
Laser inline not working at all. Output at 0.1V the whole time regardless of the S state. Used pin PA1.

HeartOfGermany, 2023-09-19:
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/12659350/Configuration_adv.txt)

I have attached the Configuration_adv.txt to the post for reference. It does not compute, may I say so? Everything works perfectly fine - all the features I enabled (and there are a few). Not however laser inline.

Just `M42 I P1 S# M1` works perfectly fine. But the delay and especially randomness makes it absolutely unusable at all.