MarlinFirmware/Marlin #26023: [BUG] EZ31865 + PT100 + BTT OCTOPUS MAX EZ

gianguardo, 2023-06-27:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hello, I'm trying to configure EZ31865 for Octopus max ez but I receive from the display err 989 temp. Can someone help me?

### Expected behavior

temp 25 (room temperature)

### Actual behavior

err 989 temp

### Steps to Reproduce

1
My setup:
Configuration.h
#define TEMP_SENSOR_0 -5
#define MAX31865_SENSOR_OHMS_0 100
#define MAX31865_CALIBRATION_OHMS_0 430
#define HEATER_0_MAXTEMP 520

Pin_BTT_OCTOPUS_MAX_EZ.h
/*
#define E1_STEP_PIN PG15 // MOTOR 6
#define E1_DIR_PIN PB3
#define E1_ENABLE_PIN PD5
#ifndef E1_CS_PIN
#define E1_CS_PIN PG9
#endif
*/
//MAX31865 MnP
#define TEMP_0_MISO_PIN PE13
#define TEMP_0_SCK_PIN PE12
#define TEMP_0_MOSI_PIN PE14
#define TEMP_0_CS_PIN PG9

Configuration_adv.h 
#define TEMP_SENSOR_FORCE_HW_SPI

Code err temp 

2
//#define TEMP_SENSOR_FORCE_HW_SPI
#define IGNORE_THERMOCOUPLE_ERRORS but i get er 989 as room temperature 

3
#define TEMP_SENSOR_FORCE_HW_SPI
#define IGNORE_THERMOCOUPLE_ERRORS 

Code err temp 

### Version of Marlin Firmware

2.1.x bug

### Printer model

Custom

### Electronics

Octopus max Ez

### Add-ons

ez31865 + pt100

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[pins_BTT_OCTOPUS_MAX_EZ.zip](https://github.com/MarlinFirmware/Marlin/files/11881389/pins_BTT_OCTOPUS_MAX_EZ.zip)
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11881390/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/11881391/Configuration_adv.zip)

ellensp, 2023-06-27:
image posted in discord  shows thermocouple is in wrong motor plug, plugged into motor 5 should be in motor 6  [image]

gianguardo, 2023-06-27:
i fixed it but i get err temp
[image]
[image]

ellensp, 2023-06-27:
you have #define software_SPI  in the pins file

this is not correct case of the keyword

must be       #define SOFTWARE_SPI

gianguardo, 2023-06-27:
It wasn't there, I added it but continue to tell me er,
The Max31865 is configured 123ON and 4Off

ellensp, 2023-06-27:
have not enabled 
#define MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED 10
or changed #define THERMOCOUPLE_MAX_ERRORS 15 to 20

gianguardo, 2023-06-27:
I changed the code, always er err

ellensp, 2023-06-27:
are you sure you actually have a PT100 ?

gianguardo, 2023-06-27:
I measured with the multimeter and I detect 112 ohm, I already tried following the manual

gianguardo, 2023-06-27:
I made some more changes:
#define MAX31865 IGNORE_INITIAL_FAULTY_READS 10
#define SHOW_TEMP_ADC_VALUES
#define M115_GEOMETRY_REPORT

Now I can read the room temperature.
I noticed another problem now as I insert the SDCard or heat the cartridge it tells me err or it reveals wrong temperatures

gianguardo, 2023-06-27:
Thanks for the help, finally now the temperatures after 200 °C tend to stabilize with an interval of 10°C.
The plate temperature sensor does not conflict with the pt100.
Inserting the SDcard immediately detects an err, how can I solve it?

ellensp, 2023-06-27:
these are support questions, not bugs.I have to close this

closed 2023-06-27 by ellensp