MarlinFirmware/Marlin #26030: [BUG] Reboot immediately after connecting via USB

Artemonim, 2023-06-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After merge commit `46b5753f5653401bc2d1d3120041a897384db48d` my Ender 3 started rebooting when trying to connect to a computer via USB. I couldn't find in which pull request this commit was accepted, but it is visible in Git

### Bug Timeline

new

### Steps to Reproduce

1. Merge `🎨 Indent temp structs` with SHA `46b5753f5653401bc2d1d3120041a897384db48d`
2. Build with Auto Build Marlin
3. Firmware the printer
4. Connect the USB cable
5. Connect to a printer via Repetier-Host V2.3.1
6. After less than 10 seconds the printer goes to reboot

### Version of Marlin Firmware

2.1.x

### Printer model

Ender-3

### Electronics

BTT SKR MINI E3 V3.0

### Bed Leveling

No Bed Leveling

### Your Slicer

None

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11894366/Configuration.zip)

thisiskeithb, 2023-06-28:
> I couldn't find in which pull request this commit was accepted, but it is visible in Git

This was directly committed by @thinkyhead.

ellensp, 2023-06-28:
there are only white space change here...  this doesn't make any sense.  C doesn't care about white spaces

Artemonim, 2023-06-28:
> @Artemonim provided configuration files are the stock BOARD_RAMPS_14_EFB configuration files Please try again

0_o
New: [Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11897433/Configuration.zip)

ellensp, 2023-06-28:
this one has MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V3_0 which matches the board you listed in the issue.

Artemonim, 2023-06-28:
Yeah, maybe I accidently switch to the main branch when I zip the files :)

thisiskeithb, 2023-06-28:
All of my printers/boards are now bootlooping once any kind of serial connection is made (TFT or USB), so I'll mark this as confirmed.

We're also discussing bootloops in https://github.com/MarlinFirmware/Marlin/issues/25852 since this issue has come and gone a couple times this month through various code cleanups. This issue may get closed in favor of that one (or both will get solved today and we can close them all together!)

thisiskeithb, 2023-06-28:
[♻️ String helper class #24390](https://github.com/MarlinFirmware/Marlin/commit/574dd34c49fde76e1e736b8a7e1ea579d724ff22) is the offending commit causing boards to reboot on serial connect.

ThomasToka, 2023-06-28:
i saw this commit and thought: wt*. if i understand all this correctly every lcd display class using serial connections will have to be adjusted. am i right?

RV-from-be, 2023-06-28:
I confirm that the printer reboot on USB. And not only, reboot also when trying to print via the SD card. This bug appeared since this commit https://github.com/MarlinFirmware/Marlin/commit/574dd34c49fde76e1e736b8a7e1ea579d724ff22
I came back before this commit and everything is back to normal.
Tested on an Ender3V2, Ender3 v1 + v427 with the jyersui as the UI.
I also tested keeping all files from this commit https://github.com/MarlinFirmware/Marlin/commit/574dd34c49fde76e1e736b8a7e1ea579d724ff22 except the dwin.cpp. Ditto reboot printers. In my opinion, the problem is either on settings.cpp, or on the marlinui.h and .cpp files, there is an overflow. But I can't trace its origin.

0xC4DE, 2023-06-29:
I am getting similar symptoms compiling for SKR V3 EZ. Let me know if there's anything I can help with. Would temp fix be to revert to before #24390 ?

ellensp, 2023-06-29:
https://github.com/MarlinFirmware/Marlin/commit/574dd34c49fde76e1e736b8a7e1ea579d724ff22 is also cause of EEPROM datasize error

RV-from-be, 2023-06-30:
@thisiskeithb @thinkyhead Shouldn't we go rollback with this commit, put it back in PR and do tests on various configurations. Adding more commits to an unstable base will make it more and more complex. It also seems to me that  PRs "at risk" should be tested longer. These are only suggestions.

thisiskeithb, 2023-07-01:
Bootlooping was fixed in https://github.com/MarlinFirmware/Marlin/pull/26037, but now temperatures are no longer reported over serial & serial devices like BTT's TFT can no longer connect.

thinkyhead, 2023-07-01:
> It also seems to me that PRs "at risk" should be tested longer. These are only suggestions.

Much appreciated. The string refactor has been sitting in the PR queue for many months, but unfortunately we don't have a group of testers standing by other than myself who will pull down PRs and test them. I can put out the word to the Testers role on Discord that something needs testing, but that has not really proven to be effective. TBH I often don't do the most thorough job of testing due to my general workload. So it is often necessary to merge a PR before it will actually get tested enough to catch all of its issues. The `bugfix-*` branch will be getting renamed to `dev` pretty soon, so hopefully that will help to clarify that the branch is not expected to be bug-free, but is expected to have egregious bugs sometimes.

thisiskeithb, 2023-07-01:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/26037 & https://github.com/MarlinFirmware/Marlin/commit/1a241e6e86512b570267d672e2f00a80df41b2e9

closed 2023-07-01 by thisiskeithb