MarlinFirmware/Marlin #26032: [BUG] RepRap full

Voxel3DPrinting, 2023-06-29:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I was compiling latest 2.1.2.1 for my duplicator 6 which I am transforming with new electronics.

After compiling and uploading the firmware, I noticed the Y axis was only moving 1 direction, I verified enstops were connected correctly, and because the Y moved in the positive direction both on + and - commands, I suspected it to be a bug.

After seeing [this post ](https://github.com/MarlinFirmware/Marlin/issues/25809) I made up my mind and downloaded the latest bugfix. I compiled it the exact same way, however, now the display is not working.

I can also not connect via USB type-B

Anybody got an Idea what it could be?

### Bug Timeline

after moving to latest bugfix

### Expected behavior

I expected the machine to boot up correctly, and working correctly (apart from tuning steps etc.

### Actual behavior

When turning on, I cannot control machine, or connect via the display nor USB port

### Steps to Reproduce

Compile  and upload the firmware on the mainboard and turn it on

### Version of Marlin Firmware

02010300, latest bugfix downloaded on 29/06

### Printer model

Duplicator 6 in its base

### Electronics

BTT SKR 3 + TMC2209 + PT100 with amplifier boards from FYSETC

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/11910196/Configurations.zip)

Voxel3DPrinting, 2023-06-29:
NOTE;

I did verify that the firmware was uploaded successfully.
After powering on, I see the statusLED flickering, and then turning off. File changed to "FIRMWARE.CUR"

This means that it indeed is uploading correctly

SwiftNick, 2023-06-29:
See bug https://github.com/MarlinFirmware/Marlin/issues/26030
There is a boot loop issue currently under investigation.

Voxel3DPrinting, 2023-06-30:
> See bug #26030 There is a boot loop issue currently under investigation.

I do not think this has to do with my particular issue, mine does not start up at all, and does not give image on the screen, nor can I connect to the machine via USB.. In the bug you mention, it boots up, but after X seconds reboots.

Is it possible to use normal marlin 2.1.2.1 and fix the Y issue? anybody ideas?

SwiftNick, 2023-06-30:
Using 2.1.2.1, if you send move instructions visa USB, does the Y axis move correctly?
Manually move to the middle of the bed then send the following GCode commands.
G91
G1 Y50 F1000
G1 Y-50

The printer should move Y axis 50mm in one direction, then 50mm in the opposite direction.

If this works ok, the issue is with your panel setup.

The USB issue you mentioned is a separate bug. Any serial connection causes an infinite boot loop, so the controller is effectively un responsive. If your panel is connected via a serial port, it too will be unresponsive.

Voxel3DPrinting, 2023-06-30:
ahh I understand, i will stick to the stable 2.1.2.1 build then.

Via USB, both movements cause the Y to move in a positive direction.

ellensp, 2023-07-02:
Issue https://github.com/MarlinFirmware/Marlin/issues/26030 has been fixed 
now this is a duplicate of https://github.com/MarlinFirmware/Marlin/issues/25809

closed 2023-07-02 by ellensp