MarlinFirmware/Marlin #26038: [BUG] (`HOST_STATUS_NOTIFICATIONS` doesn't compile without a display anymore)

jounathaen, 2023-07-01:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi,

When I enable `HOST_STATUS_NOTIFICATIONS`, but don't enable any configuration that enables some kind of display, the compilation fails.

That was most probably introduced in 574dd34c49f, which enables the compilation of `MarlinUI::host_status()` in any circumstance, but the variable `status_message` is only available, if `HAS_DISPLAY` is set.

### Bug Timeline

Probably since 574dd34c49f

### Expected behavior

Successful compilation :stuck_out_tongue: 

### Actual behavior

Compilation fails with the following error:

```
In file included from Marlin/src/lcd/../inc/MarlinConfigPre.h:37:0,
                 from Marlin/src/lcd/../inc/MarlinConfig.h:28,
                 from Marlin/src/lcd/marlinui.cpp:23:
Marlin/src/lcd/marlinui.cpp: In static member function 'static void MarlinUI::host_status()':
Marlin/src/lcd/marlinui.cpp:1435:50: error: 'status_message' was not declared in this scope
   TERN_(HOST_STATUS_NOTIFICATIONS, hostui.notify(status_message));
                                                  ^
Marlin/src/lcd/../inc/../core/macros.h:607:26: note: in definition of macro 'THIRD'
 #define THIRD(a,b,c,...) c
                          ^
Marlin/src/lcd/../inc/../core/macros.h:206:29: note: in expansion of macro '___TERN'
 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
                             ^~~~~~~
Marlin/src/lcd/../inc/../core/macros.h:205:29: note: in expansion of macro '__TERN'
 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
                             ^~~~~~
Marlin/src/lcd/../inc/../core/macros.h:204:29: note: in expansion of macro '_TERN'
 #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION ? 'A' : '<nul>'
                             ^~~~~
Marlin/src/lcd/marlinui.cpp:1435:3: note: in expansion of macro 'TERN_'
   TERN_(HOST_STATUS_NOTIFICATIONS, hostui.notify(status_message));
   ^~~~~
Marlin/src/lcd/marlinui.cpp:1435:50: note: suggested alternative: 'status_printf'
   TERN_(HOST_STATUS_NOTIFICATIONS, hostui.notify(status_message));
                                                  ^
Marlin/src/lcd/../inc/../core/macros.h:607:26: note: in definition of macro 'THIRD'
 #define THIRD(a,b,c,...) c
                          ^
Marlin/src/lcd/../inc/../core/macros.h:206:29: note: in expansion of macro '___TERN'
 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
                             ^~~~~~~
Marlin/src/lcd/../inc/../core/macros.h:205:29: note: in expansion of macro '__TERN'
 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
                             ^~~~~~
Marlin/src/lcd/../inc/../core/macros.h:204:29: note: in expansion of macro '_TERN'
 #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION ? 'A' : '<nul>'
                             ^~~~~
Marlin/src/lcd/marlinui.cpp:1435:3: note: in expansion of macro 'TERN_'
   TERN_(HOST_STATUS_NOTIFICATIONS, hostui.notify(status_message));
   ^~~~~
```

### Steps to Reproduce

1. activate `HOST_ACTION_COMMANDS`
2. activate `HOST_PROMPT_SUPPORT`
3. activate `HOST_STATUS_NOTIFICATIONS`
4. run `pio run -c platformio.ini`

### Version of Marlin Firmware

bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Minimal example configuration:
[Configuration-host-status-notifications.zip](https://github.com/MarlinFirmware/Marlin/files/11926376/Configuration-host-status-notifications.zip)

ellensp, 2023-07-02:
to me this is just a invalid setting

Configuration_advance should be updated to be
 
  #if ALL(HOST_PROMPT_SUPPORT, HAS_STATUS_MESSAGE)
    #define HOST_STATUS_NOTIFICATIONS   // Send some status messages to the host as notifications
  #endif
so you cannot enable  HOST_STATUS_NOTIFICATIONS when there is no HAS_STATUS_MESSAGE

thisiskeithb, 2023-07-02:
You can enable `HOST_ACTION_COMMANDS` / `HOST_PROMPT_SUPPORT` without an LCD, so it'd be good to fix `HOST_STATUS_NOTIFICATIONS` so it can also work without an LCD enabled. Then, serial-connected hosts like OctoPrint can display messages from Marlin.

I thought this was working at one point, but I've probably always had an LCD enabled when running OctoPrint.

ellensp, 2023-07-02:
See https://github.com/MarlinFirmware/Marlin/pull/22833

"This adds the host notification to the 3rd method mentioned above and adds a configuration option: HOST_STATUS_NOTIFICATIONS. If enabled, any time the UI status message is updated using any of the 3 functions mentioned above, it will also be echoed back as an action:notifiication. Without that enabled, only specific conditions trigger a notification."

With no UI this is an invalid selection

thisiskeithb, 2023-07-02:
These should probable also get wrapped with `HAS_STATUS_MESSAGE` like they were before 574dd34:

https://github.com/MarlinFirmware/Marlin/blob/3ea87ad54c1e445a1c8cfef1489809edae5decb3/Marlin/src/lcd/marlinui.cpp#L1428-L1436

thisiskeithb, 2023-07-02:
`HAS_STATUS_MESSAGE` can't be used in the configs since it's defined later, but @GMagician submitted a fix in https://github.com/MarlinFirmware/Marlin/pull/26040.

PR #26040 (2023-07-05): Fix HOST_STATUS_NOTIFICATIONS compile error
Fix #26038
closed 2023-07-02 by thisiskeithb