MarlinFirmware/Marlin #26039: [BUG] Temperature reporting is broken after #24390 and #26037, no connection with BTT TFT

rondlh, 2023-07-01:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Temperature reporting is completely missing after #24390 and #26037.

The BTT TFT touch screens use the temperature reports to confirm the connection with the motherboard. If the temp reports are not coming then the TFT will report that "No printer connected" and will not work.

### Bug Timeline

2023-06-30

### Expected behavior

Temp reporting like before

### Actual behavior

No temperature reporting at all

### Steps to Reproduce

1. Issue an M155 S1
2. Check if the temperature is reported every second

### Version of Marlin Firmware

Latest Marling 2.1.x bugfix 2023-07-01

### Printer model

Custom

### Electronics

MKS Monster8 V1.0

### Add-ons

BTT TFT35 V3.0

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/11926828/Configuration_adv.zip)

thisiskeithb, 2023-07-01:
This has been confirmed in the comments of https://github.com/MarlinFirmware/Marlin/pull/26037

EEPROM is also broken for certain builds as well.

thisiskeithb, 2023-07-01:
Here's a temperature reporting workaround from @RV-from-be: https://github.com/MarlinFirmware/Marlin/pull/26037#issuecomment-1616054195

In `temperature.cpp`, modify:
https://github.com/MarlinFirmware/Marlin/blob/4985acafada69dcb17b813d76b359bcfb5b57b43/Marlin/src/module/temperature.cpp#L4184

to:
```cpp
    #ifdef SERIAL_FLOAT_PRECISION
      #define SFP _MIN(SERIAL_FLOAT_PRECISION, 2)
    #else
      #define SFP 2
    #endif
```
and:

https://github.com/MarlinFirmware/Marlin/blob/4985acafada69dcb17b813d76b359bcfb5b57b43/Marlin/src/module/temperature.cpp#L4188

to:
```cpp
    s += F(":"); s += p_float_t(c, SFP);
```

thisiskeithb, 2023-07-01:
Fixed in https://github.com/MarlinFirmware/Marlin/commit/1a241e6e86512b570267d672e2f00a80df41b2e9

closed 2023-07-01 by thisiskeithb