MarlinFirmware/Marlin #26044: [FR] Octopus Pro v1.1 (H723ZE) no definition

vfbank, 2023-07-03:
### Is your feature request related to a problem? Please describe.

There is no definition of modern hardware.

### Are you looking for hardware support?

Octopus Pro v1.1 (H723ZE)

### Describe the feature you want

Even in the latest version of the Marlin firmware, there is no definition to support the H723ZE chipset in the boards.h. How do I use it?

thisiskeithb, 2023-07-03:
Added in https://github.com/MarlinFirmware/Marlin/pull/26043

closed 2023-07-03 by thisiskeithb

vfbank, 2023-07-03:
> #26043에 추가됨

Compilation fails. Using the old Octopus Pro 1.0 definitions results in an STM32F4 error, and using the Octopus Pro Max definitions compiles but crashes the firmware.

thisiskeithb, 2023-07-03:
> Compilation fails. Using the old Octopus Pro 1.0 definitions results in an STM32F4 error, and using the Octopus Pro Max EZ definitions compiles but crashes the firmware.

You don't reuse any of the Octopus Pro 1.0 / STM32F4 definitions with the referenced PR.

Use the files from [the PR's source fork](https://github.com/thisiskeithb/Marlin/tree/pr/add_octopus_pro_1_1_stm32H723zet6) or wait until it is merged.

Using a minimal config, this compiles fine.

<details><summary>Minimal Config:</summary>
<p>

```diff
diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index b21eba7..2c7199f 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -67,7 +67,7 @@
 
 // Choose the name from boards.h that matches your setup
 #ifndef MOTHERBOARD
-  #define MOTHERBOARD BOARD_RAMPS_14_EFB
+  #define MOTHERBOARD BOARD_BTT_OCTOPUS_PRO_V1_1
 #endif
 
 /**
@@ -78,7 +78,7 @@
  *
  * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
  */
-#define SERIAL_PORT 0
+#define SERIAL_PORT -1
 
 /**
  * Serial Port Baud Rate
diff --git a/platformio.ini b/platformio.ini
index e3bdb6f..f54fee7 100644
--- a/platformio.ini
+++ b/platformio.ini
@@ -13,7 +13,7 @@
 [platformio]
 src_dir      = Marlin
 boards_dir   = buildroot/share/PlatformIO/boards
-default_envs = mega2560
+default_envs = STM32H723ZE_btt
 include_dir  = Marlin
 extra_configs =
     Marlin/config.ini
```

</p>
</details>

Compile log using the above minimal config:

<details><summary>STM32H723ZE_btt  SUCCESS:</summary>
<p>

```prolog
Archiving .pio/build/STM32H723ZE_btt/libFrameworkArduino.a
Indexing .pio/build/STM32H723ZE_btt/libFrameworkArduino.a
Linking .pio/build/STM32H723ZE_btt/firmware.elf
Checking size .pio/build/STM32H723ZE_btt/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   2.0% (used 11544 bytes from 577536 bytes)
Flash: [==        ]  20.0% (used 104916 bytes from 524288 bytes)
Building .pio/build/STM32H723ZE_btt/firmware.bin
====================================== [SUCCESS] Took 40.74 seconds ======================================

Environment      Status    Duration
---------------  --------  ------------
STM32H723ZE_btt  SUCCESS   00:00:40.735
======================================= 1 succeeded in 00:00:40.735 =======================================
 *  Terminal will be reused by tasks, press any key to close it. 
```

</p>
</details>

Hman32, 2023-08-06:
Hi @thisiskeithb  there is also a Octopus Pro 1.0.1 with the H723 so, I assume the PIN is not the same as the PRO V1.1 H723
Do I need to create a new "issue" for that one ?

Thank's

thisiskeithb, 2023-08-06:
> there is also a Octopus Pro 1.0.1 with the H723 so, I assume the PIN is not the same as the PRO V1.1 H723

Do you have a photo of the board or listing? BigTreeTech doesn't list anything in the Octopus Pro repo for a "1.0.1" version.

Hman32, 2023-08-06:
@thisiskeithb thanks for the quick reply, there some picture of my board:

[image]

[image]

[image]

Hope those will help.

thisiskeithb, 2023-08-06:
> Hope those will help.

Thanks! Give the Octopus Pro 1.1 board define & environment a try and see if it works.

Hman32, 2023-08-06:
@thisiskeithb  will do, but this board has the issue with the DFU and the fix they have done for the V1.1 changed some PIN's.
So it will compile but I will not be able to use it if the there is some deviation?

<html>
<body>

Title | Octopus Pro V1.0/V1.0.1 | Octopus Pro V1.1
-- | -- | --
HE0 | PA2 | PA0
HE2 | PB10 | PB0
Motor4-EN | PA0 | PA2
RGB | PB0 | PB10

</body>
</html>

thisiskeithb, 2023-08-06:
> Edit: at least I can also use the V1.1 and change the PIN to revert the above change :D

That will work for now. I'm prepping a PR that will add support for this 1.0.1 board.