MarlinFirmware/Marlin #26058: [BUG] OCTOPUS MAX EZ SDCARD NOT RECOGNIZED

gianguardo, 2023-07-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi I am having this problem with my octopus max ez can anyone help me please
1 cannot read the sdcard in the internal slot 
2 if I connect it to the LCD display it recognizes it but as I try to print the card restarts
3 i tried to use a usb connection but i have the same error as the lcd

### Version of Marlin Firmware

2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11971378/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/11971379/Configuration_adv.zip)

gianguardo, 2023-07-11:
Update:
Using the latest version of marlin now I can print via lcd (after initializing EPPROM as I receive temp err) and via usb but if I use the SDcard via the card it doesn't detect it for me.

ellensp, 2023-07-11:
One potential issue I see

You have TMC5160 which uses  SPI

//
// Default pins for TMC software SPI
//
#ifndef TMC_SPI_MOSI
  #define TMC_SPI_MOSI                      PE14
#endif
#ifndef TMC_SPI_MISO
  #define TMC_SPI_MISO                      PE13
#endif
#ifndef TMC_SPI_SCK
  #define TMC_SPI_SCK                       PE12
#endif

but the SDCARD uses same SPI but needs hardware

you have #define TMC_USE_SW_SPI in Configuration._adv.h 

Disable this

gianguardo, 2023-07-11:
Hi, I tried disabling TMC_USE_SW_SPI in Configuration._adv.h but it still doesn't detect my sdcard.

ellensp, 2023-07-11:
you set #define SDCARD_CONNECTION ONBOARD in Configuration_adv.h to make it use the onboard sdcard

gianguardo, 2023-07-11:
Now it works, thank you very much for the help

closed 2023-07-11 by gianguardo