MarlinFirmware/Marlin #26059: AttributeError: 'dict' object has no attribute 'get_debug_tool_name'

Jonas-Alber, 2023-07-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

No debugging possible with Marlin projects in VSCode.
AttributeError: Traceback (most recent call last):
  File "C:\Users\uic98235\.platformio\penv\Lib\site-packages\platformio\__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
    ^^^^^
  File "C:\Users\jonas\.platformio\penv\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\jonas\.platformio\penv\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\jonas\.platformio\penv\Lib\site-packages\platformio\cli.py", line 85, in invoke
    return super().invoke(ctx)
           ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\jonas\.platformio\penv\Lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\jonas\.platformio\penv\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\jonas\.platformio\penv\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\jonas\.platformio\penv\Lib\site-packages\click\decorators.py", line 34, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\jonas\.platformio\penv\Lib\site-packages\platformio\debug\cli.py", line 104, in cli
    debug_config = _configure(*configure_args)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\jonas\.platformio\penv\Lib\site-packages\platformio\debug\cli.py", line 115, in _configure
    debug_config = DebugConfigFactory.new(
                   ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\jonas\.platformio\penv\Lib\site-packages\platformio\debug\config\factory.py", line 34, in new
    board_config.get_debug_tool_name(
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'dict' object has no attribute 'get_debug_tool_name'

### Bug Timeline

Today

### Expected behavior

Start Debugger

### Steps to Reproduce

Unkown

### Version of Marlin Firmware

2.1.2.1

### Electronics

STM32F407ZET6

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

The problem occurs both in new projects with the current Marlin-bugfix-2.1.x and in existing projects.
According to Platformio Support, the problem is with Marlin.
Testing with a Platformio STM32 setup problem confirms that debugging works in principle. The problem occurs only in Marlin projects.
Error occurs as well with Configuration of other Printers like ANETET4 on a STM32F407VGT6

ellensp, 2023-07-07:
get_debug_tool_name is part of penv/lib/python3.10/site-packages/platformio/platform/board.py  as provided by platformio

You need to provide way more details.

Ie a step by step to reproduce

Jonas-Alber, 2023-07-10:
Ich kenne mich viel zu wenig mit Platformio und Marlin aus um den Fehler zu reproduzieren. Da das Problem bei sämtlichen Marlin Projekten auftritt kann ich auch nicht auf ein Funktionierendes Projekt zurückgreifen.
Was ich bisher herausgefunden habe ist, dass die env_name simulator_linux_debug ist und das Folgende der Inhalt der board_config, bei welcher die get_debug_tool_name ihren fehler wirft
cmd```
{'marlin_Artillery_Ruby': <platformio.platform.board.PlatformBoardConfig object at 0x000002E33D804E50>, 'marlin_BTT_SKR_SE_BX': <platformio.platform.board.PlatformBoardConfig object at 0x000002E3408EB510>, 'marlin_BigTree_BTT002': <platformio.platform.board.PlatformBoardConfig object at 0x000002E340545150>, 'marlin_BigTree_BTT002_VET6': <platformio.platform.board.PlatformBoardConfig object at 0x000002E34093B610>, 'marlin_BigTree_GTR_v1': <platformio.platform.board.PlatformBoardConfig object at 0x000002E3409226D0>, 'marlin_BigTree_Octopus_Pro_v1_F429': <platformio.platform.board.PlatformBoardConfig object at 0x000002E340850310>, 'marlin_BigTree_Octopus_v1': <platformio.platform.board.PlatformBoardConfig object at 0x000002E3408C9310>, 'marlin_BigTree_SKR_Pro': <platformio.platform.board.PlatformBoardConfig object at 0x000002E3408C9450>, 'marlin_CREALITY_STM32F401RC': <platformio.platform.board.PlatformBoardConfig object at 0x000002E3408C93D0>, 'marlin_FYSETC_CHEETAH_V20': <platformio.platform.board.PlatformBoardConfig object at 0x000002E3408C9510>, 'marlin_MKS_SKIPR_V1': <platformio.platform.board.PlatformBoardConfig object at 0x000002E340947D90>, 'marlin_STM32F401RC': <platformio.platform.board.PlatformBoardConfig object at 0x000002E340945910>, 'marlin_STM32F407VET6_CCM': <platformio.platform.board.PlatformBoardConfig object at 0x000002E340946310>, 'marlin_STM32F407VGT6_CCM': <platformio.platform.board.PlatformBoardConfig object at 0x000002E340729D50>, 'marlin_STM32F407ZET6': <platformio.platform.board.PlatformBoardConfig object at 0x000002E340944110>, 'marlin_STM32F407ZGT6': <platformio.platform.board.PlatformBoardConfig object at 0x000002E340946E10>, 'marlin_STM32F429VGT6': <platformio.platform.board.PlatformBoardConfig object at 0x000002E340955A10>, 'marlin_STM32F446ZET_tronxy': <platformio.platform.board.PlatformBoardConfig object at 0x000002E340954D50>, 'marlin_STM32G0B1RE': <platformio.platform.board.PlatformBoardConfig object at 0x000002E34095ED50>, 'marlin_STM32H743Vx': <platformio.platform.board.PlatformBoardConfig object at 0x000002E34095F810>, 'marlin_archim': <platformio.platform.board.PlatformBoardConfig object at 0x000002E34095DF10>, 'marlin_blackSTM32F407VET6': <platformio.platform.board.PlatformBoardConfig object at 0x000002E34096AE10>, 'marlin_fysetc_s6': <platformio.platform.board.PlatformBoardConfig object at 0x000002E34096AD50>, 'marlin_maple_CHITU_F103': <platformio.platform.board.PlatformBoardConfig object at 0x000002E34093E590>, 'marlin_maple_MEEB_3DP': <platformio.platform.board.PlatformBoardConfig object at 0x000002E34096B910>, 'marlin_opulo_lumen_rev3': <platformio.platform.board.PlatformBoardConfig object at 0x000002E340970FD0>, 'marlin_opulo_lumen_rev4': <platformio.platform.board.PlatformBoardConfig object at 0x000002E340971E90>}
```
Can you give me instructions on how I could best reproduce the error?

ellensp, 2023-07-10:
you cannot use this board in the simulator

to use simulator_linux_debug you must set #define MOTHERBOARD BOARD_SIMULATED

Jonas-Alber, 2023-07-10:
I do not want to use the simulator but the specified board.

ellensp, 2023-07-10:
then env_name should be set to what your board needs

which is unknown as you have created a custom board.
 all we have is the name BOARD_SCARA_CONTROLLER

Jonas-Alber, 2023-07-10:
Ok, entschuldige dass ich deine Zeit verschwendet habe. Ich habe inzwischen herausgefunden, dass ich einen Fehler bei der  Konfiguration der Debugeinstellungen in VSCode gemacht habe. Anscheinend haben sich die Einstellungen der Debugger Datei in der Entwicklung (warscheinlich durch mich) auf ihre Standardwerte zurückgesetzt und ich habe es nicht gemerkt.
Vielen Dank für deine Hilfe.

closed 2023-07-10 by Jonas-Alber