MarlinFirmware/Marlin #26062: [BUG] (G30 not working)

bart-tech3d, 2023-07-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Single probe not working after command G30

### Bug Timeline

After compile Marlin 2.1.

### Expected behavior

After command G30 go Z up and probe i deployed. Then Z go down to measure bed.

### Actual behavior

After command G30 nothing happens at all.

### Version of Marlin Firmware

Marlin-2.1.2.1 

### Printer model

own printer - Cartesian

### Electronics

MKS robin nano V3

### Add-ons

TMC2209, Dual Z endstop, Dual Z stepper motors, BL-TOUCH

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/11985753/Marlin.zip)

ThomasToka, 2023-07-08:
I have it in my fork included for manual or automatic measuring the tramming points and it still works.
ex
```
Send: G30 X117.5 Y117.5
[...]
Recv: Bed X:117.5 Y:117.5 Z:0.020 (i have extended it to 3 decimals, as default are two..)
```

bart-tech3d, 2023-07-10:
...by the way, I use this function to define and correct the probe offset Z, e.g. after changing the nozzle before autocalibration.
However, I changed the function so that after the measurement the offset is saved to a negative value as from probe offset. I tried the probe offset wizard, but it's unnecessarily complicated - keep pointing and flying through the menu here and there. I wrote a calibration gcode where everything is included (find offset to autoleveling) and at the end the mesh is stored in the eeprom. Very user-friendly.

closed 2023-09-19 by github-actions[bot]