MarlinFirmware/Marlin #26063: [FR] (function to find the z-probe offset )

bart-tech3d, 2023-07-07:
### Is your feature request related to a problem? Please describe.

I need to simply find the z-probe offset for BL-TOUCH without having to convert and add values.

I added the function G36 , which is based on the principle of G30, except that the measured values are negated and saved as z-probe offset.
The measured value is corrected to the paper during the sequence (the distance between the nozzle and the pad). In the new marlin 2.1. unfortunately this no longer works, nor does the G30 function itself - it does not respond.

This function will greatly facilitate leveling, especially when changing the nozzle.

### Are you looking for hardware support?

own cartesian printer with dual z-motors and endstops and BL-TOUCH probe

### Describe the feature you want

1. after starting the leveling sequence, the printer is referenced by independent dual (or single) limit switches
(intentionally the X-axis is distorted so that the alignment can be seen)
2. Aim the pad and drive the nozzle to the same measuring point at Z0
3. the printer waits for a press to continue
4. after pressing, the 40s time needed to set the offset on the paper runs (it would be better to confirm again with the user)
5. the printer itself completes the calibration and stores the right mesh values in the EEPROM memory
[G36.zip](https://github.com/MarlinFirmware/Marlin/files/11986176/G36.zip)

### Additional context

My own leveling sequence in Marlin 2.0. See video below.
https://youtu.be/stuhpdqXV1E

thisiskeithb, 2023-07-07:
You can use `PROBE_OFFSET_WIZARD` to find the Z probe offset:

https://github.com/MarlinFirmware/Marlin/blob/bcf61cc91c00b2c0d82aa0e88d0f88e5a18bbe8b/Marlin/Configuration_adv.h#L1478

closed 2023-07-07 by thisiskeithb

bart-tech3d, 2023-07-07:
OK, thank you very much 👍

bart-tech3d, 2023-07-10:
I tried and it works. However, compared to the complete calibration sequence, which includes finding and correcting the offset via autoleveling and saving it to the eeprom, the wizard is user-complicated for me. Keep pointing and flying in the menu here and there.
In my opinion, finding and correcting the z-probe offset is an integral part of the pad calibration - for example, when changing the nozzle. It's a shame that no one wants to deal with it, it would certainly help.
E.g. Klipper already has it implemented in the autoleveling function.

I think it would be enough to create a function by merging wizard+autoleveling+save to eeprom and nothing more.

EvilGremlin, 2023-07-10:
BABAYSTEP_ZPROBE_OFFSET is superior, much easier and faster, method.